arduino/Arduino #2896: links from Arduino site not working - wifi shield

alexmac131, 2015-04-03:
https://github.com/arduino/Arduino/tree/master/hardware/arduino/firmwares

Does not work http://arduino.cc/en/Hacking/WiFiShieldFirmwareUpgrading

alexmac131, 2015-04-27:
The issue also relates to the removal of the firmware tools in the current master branch.  One possible fix I am going to do is go back to a revision that has the bits used by the firmware hardware/tools/  sub dir and give it a go.

alexmac131, 2015-04-28:
So here is the rather odd little fix...

Check out the base Arduino code  for  ide-1.0.x,  copy in the missing the tools into your current release. I am use 1.6.2 for the hardware, you have to change the mac fix script and you need to copy in the wifi firmware as well

agdl, 2015-05-21:
The correct link is https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/firmwares I updated the on-line page!

closed 2015-05-21 by agdl