arduino/Arduino #2899: 1.6 upgrade concern on OSX

vikfors, 2015-04-03:
[image]
Compiling the same code on 1.0.5 and 1.6.3 gives the results shown
What is wrong?
Leif

PaulStoffregen, 2015-04-03:
You have to add "const" to the variable.  The old avr-gcc toolchain didn't require this.  The new one does.

ffissore, 2015-04-04:
@vikfors see https://github.com/arduino/Arduino/wiki/1.6-Frequently-Asked-Questions#errors-related-to-avr-progmem-changes

vikfors, 2015-04-04:
OK,
Thanks for the info.
In my case changing the variable from static to const did solve the problem.
Tested and that works fine.
I also had a problem with a project that was using SD.h
This was calling SPI.h
That was resolved by including SPI.h in the project directly.
A bit weird that it no longer works when called indirectly and it took me a while to find the solution.

closed 2015-04-04 by vikfors