arduino/Arduino #2901: Default colors for errors is illegible

Coding-Badly, 2015-04-04:
There are numerous complaints about the dark-red text on black background colors for error messages.  For good reason.  The error messages are ILLEGIBLE.

rogerclarkmelbourne, 2015-04-04:
I agree. I have to highlight them to read them. Dark red on a black background is terrible UI design

mschlenker, 2015-04-04:
How about the possibility to set some colors in some config file or as an environment variable? This way we could paly around with some color combinations before settling? 

BTW: I am making lots of screenshots for books, there dark text on light background looks much better than the other way round, especially in books printed in black and white.

PaulStoffregen, 2015-04-04:
> How about the possibility to set some colors in some config file

Like **lib/theme/theme.txt** ?

Coding-Badly, 2015-04-04:
Paul, thank you.  I could not remember the filename and searching through Google only led me to preferences.txt.

Coding-Badly, 2015-04-04:
In my opinion, the saturation for #A61717 is too low.  Making it bright enough to be visible washes out the color.  #FF1717 is much more legible.

PaulStoffregen, 2015-04-04:
For Teensy, I'm patching it to be #FF6060, which looks really good.

mschlenker, 2015-04-04:
Why not go with #FFFF00? 

I basically agree with #FF6060. The bikeshed has a nice new color now.

PaulStoffregen, 2015-04-04:
Everyone can paint their lib/theme/theme.txt bikeshed any color they like.

PaulStoffregen, 2015-04-04:
But you did ask why.  Well, in case you're not familiar with Teensy, I publish an installer which adds quite a large number of patches to Arduino.  Teensy has quite a good number of features that aren't supported by the stock Arduino software, particularly relating to non-Serial USB modes.

My general policy it to avoid changing the IDE's behavior for non-Teensy boards, even when it might be beneficial.  Instead, I take the long road of contributing improvements to Arduino which eventually become part of the official IDE.  But sometimes, like this color change, adding code to preserve the original color when a non-Teensy board is selected is just too much work.

So I chose the brightest color that still has some resemblance to the original.  Changing it to bright yellow might be even better from a readability point of view.  But from a "why did your installer change the color" point of view, that's a pretty dramatic change that shouldn't be done by an installer like mine that modifies a user's existing copy.

mschlenker, 2015-04-04:
Paul, I completely agree with you. I think it's best to use red to indicate error - and then, choose a bright red. For personal settings like mine, reverse colors (for the books) or bright yellow or bright orange on black (working in the night) probably will do the job.

I am currently working on some Yún related project that probably will have some trickle down effects on Arduino (.cc's) Yún shield. And there some questions of design choice raise. And every suggestion has to be made carefully not breaking compatibility to older firmware and Doghunter boards. So discussing colors is just fun.

Thanks for hinting me to the appropriate location!

00alis, 2015-04-07:
We reverted the Error color to the original one, see 8020c0d7339ede208f4a60934e65efb9a52b63df.
Fix available in the Nightly Build from tomorrow on, will be included in the next release.

Thanks for your feedback!

closed 2015-04-07 by ffissore

Coding-Badly, 2015-04-07:
Huh.  A burnt orange.  You must be a Longhorn fan.

That is much better.  Thanks.

lmihalkovic, 2015-11-21:
see #4151 / #4158