arduino/Arduino #2905: Strange behavior in 1.6.2, upload stucks

fhpa, 2015-04-04:
/*
The following nonsense program is the rest of a big program
which i suddenly could not upload anymore to a Mega 2560.

So i broke it down to the following rest lines. These
still i could not uload to a Mega 2560

But if i out-comment the line 19 it will be possible to upload

Also if i remove one of the exclamation marks in line 28 i can
upload it to the board

I did not try it with another kind of Arduino board.

I am using IDE 1.6.2, german version

*/

String tosend;

void setup(){
}

void loop(){
}

void requestEvent(){
Serial.println("!!!");
}

Lauszus, 2015-04-04:
This has been fixed already. Please see: https://github.com/arduino/Arduino/pull/1183. Please try to update the bootloader on your Mega 2560.

PaulStoffregen, 2015-04-04:
Did you buy a cheap Chinese clone, or a genuine Arduino Mega?

fhpa, 2015-04-04:
Hi Paul.

thank you for answering!

I tried it with bot original and clone. It was the same behavior.

I tried it on either IDE 1.6.2 and 1.6.3 - it was always the same.

Later i will try to update the bootloader...

best regards
franz

fhpa, 2015-04-04:
Hi Kristian,

thank you for answering!

I will try to update the bootloaders. They are on date of last year...

best regards
franz

ffissore, 2015-04-30:
Closing for lack of feedback

closed 2015-04-30 by ffissore