arduino/Arduino #2909: Arduino 1.6.3 windowbug on Mac OS

chq76, 2015-04-04:
Hi,

when opening Arduino 1.6.3 it always opens in a small window in the lower left corner.

13" MacBook Air, Mac OS 10.10.2

Greetz Chris

akkoyun, 2015-04-12:
i get the same bug.

13" Macbook pro 10.10.3

wabbitguy, 2015-04-23:
Make that three of us. I also noticed that the when you edit a sketch the green window jelly top right corner doesn't mark the window as dirty. I'm still trying to figure out why the ports in use are now /dev/cu instead of /dev/tty...oh well.

chq76, 2015-04-23:
Arduino 1.6.3 feels buggy on Mac!

Greetz Chris

ffissore, 2015-05-29:
Do you have tried 1.6.4 yet? Is it affected by the problem? If so, can you please give a try to the latest hourly build? http://www.arduino.cc/en/Main/Software#hourly We closed many issues, some of fixes may be related to your issue

wabbitguy, 2015-05-30:
I just tried the May 29th build. Same issue so no change that I can see.

1. Shut down the IDE with a full window sketch showing.
2. Run the IDE again
3. IDE reopens the sketch at proper size and then immediately shrinks it down to the bottom left corner like 300 x200 pixels in size.

If you have several sketches open, the IDE doesn’t always remember them, and it gets confused as to which one was the front most when it quit. One of the windows will still end up shrunk in the bottom left corner.

With two sketches open, and you close one, IDE becomes confused and will alert you that if you close this “last” window it will quit. Except there are two sketches open, not one.

If you paste a hardware folder inside the apps folder, it will screw up the permissions and OS X will refuse to run the program. OS X says it’s damaged. However if you run the program once, then you can paste in stuff without that happening. I was pasting in DigiSpark hardware when I found this one. There’s probably a board manager or a proper way to do it, but since documentation is scant and the help file pretty much useless, well, paste works…LOL

Mel
---

ffissore, 2015-06-01:
@wabbitguy so far I tried to *imagine* the bug. Can you please provide a video?

wabbitguy, 2015-06-01:
Unfortunately, I don’t have to imagine them, I see them every time I run the brain dead IDE..LOL

I’ll see if I have some screen recording software here and do up some vids for you since that’s about the only way I can make a video. I’ll try and get to it in the next day or so.

Mel
---

damellis, 2015-06-04:
This looks related: https://community.oracle.com/thread/2518453

wabbitguy, 2015-06-05:
That might explain the windows resizing, but it sure doesn’t go to any length to explain:

1. why the IDE forgets opened sketches, something wrong because the paths are in the prefs it saves.

2. when you use File Menu item QUIT, the IDE pops up a dialog that says closing the last window will exit the IDE. Apparently the IDE doesn’t know the difference between a MENU event and a window closing event. Really?

3. Abandon the sketch tab dirty marker theme. Use the window’s red close jelly and mark it as “dirty” since it’s been done that way for the last 15 years and that’s what we’re used to looking for in a GUI. If we have to get saddled with something non-standard like the tab dirty markers (to indicate a sketch has been modified), at least pick something that doesn't look like a hieroglyphic character. Difficult to see. Use something like OPT G for a dot or something that stands out.

If only one sketch was open, the IDE randomly forgets it and reloads the default “new” blank sketch on the next run. And on the “new” blank sketch the IDE manages to put that in the middle of the monitor.

if I have 4 or 5 sketches open, when the IDE starts up, they all show in the original full window size I last used and then, one by one, they get shrunk down to the bottom left corner of the display. How hard is it to remember the TLBR of the window and put it back there. It’s. Not. That. Hard.

Difficult to blame the JAVA SDK for all of this.

I’m using the May 28th build. I’ll try the June 5th one, if there’s a list of what has been “fixed”, it’s a little too well hidden. That should be included with the build so we can directly test the fixes. Not sure that hourly “builds” are the best way to do testing any way. Sigh.

Mel
---

wabbitguy, 2015-06-05:
Nightly, hourly, by the minute…or whatever. June 5, 10:15:8…

The dialog that appeared when using the Quit item (one that warned about quitting the iDE if you close the last window) doesn’t show any more. Thus it now works correctly.

No more randomly forgetting open sketches. So far so good.

I like the File, Open Previous item!

Not fixed:

Windows still resizing to bottom left corner on open.
Window close button not marked as “dirty” on window contents change. Same silly character appears in the tab header. Non-standard GUI abuse. Should be something like “Window.contentschanged = true”.

Mel
---

ffissore, 2015-06-19:
I can't reproduce it. It would be nice to have some help from a developer using macosx.

damellis, 2015-06-19:
I see the same problem in 1.6.5: maximize a sketch window, quit Arduino, and restart Arduino. The sketch starts maximized and immediately resizes to just the lower-right corner (see screenshot). This appears to the be the minimum size of the window (i.e. it's the same size as the smallest I can make the window by manually resizing it). The resizing animation is the same (I think) as what you see when you click the green (maximize / restore) button. It's almost as if Java is deciding that the window was maximized, but isn't any more, and so it should restore it to the "unmaximized" size and location. 

[image]

ffissore, 2015-06-22:
Thank you @damellis that explains the problem. I should have fixed it with proper hadling of maximized state. Try next hourly build http://www.arduino.cc/en/Main/Software#hourly

commit f255319f (2015-06-22): Maximized state of IDE gets properly stored and restored. Fixes #2909
closed 2015-06-22 by ffissore