arduino/Arduino #2912: Version number too low, please raise to 2.0.0

mschlenker, 2015-04-04:
"The other guys and gals™" just released 1.7.0, judging features based on "This side©" 1.5.x or 1.6.0. Thus I suggest racing, ahemmm, raising the version number accordingly.

PaulStoffregen, 2015-04-04:
In the long run, copying older versions of Arduino and falsely labeling them with a higher versions is only going to destroy their credibility with the few people who can overlook all the other shady stuff they've done.

mschlenker, 2015-04-05:
What makes matters worse is that their github repos README.md points here for issues.

sej7278, 2015-04-06:
Please don't mess with versions just to beat the "other side".

cmaglie, 2015-04-07:
We are not going to change version because a fork of the Arduino IDE has been re-released with an upper version number.

Closing as wontfix.

closed 2015-04-07 by cmaglie

matthijskooijman, 2015-04-07:
@cmaglie, FWIW, it might be sane to consider to skip the 1.7.x version later on, to prevent confusion. But that's something to consider when the a new minor version number is needed, and the dust has settled wrt the .org situation.