arduino/Arduino #2914: re: poor compiler/source code error reporting

jbhome, 2015-04-05:
I was not sure where to put this post. Please feel free to move it if you have a better place.

First things first. I would like to complement you on an amazing piece of work.
I am old enough to remember the first wave of embedded micros, the 6800, 8080, and 6502, and the cost and capability of the kits.
The Arduino is so much better in so many ways.

Now the reason for this post.
This is more a compiler  issue but... here goes.
Sorry to be long winded but...
I ran OS2 Warp for a while and kept getting a long 16 + digit hex error message. When I went to the OS2 club and commented on it, one of the gurus commented, run program x, enter the multidigit code and it will tell you what the error is. In this case, there is not floppy in the disk.
So back to Arduino in particular but C compilers in general.
The errors produces are typically vague and usually mean your missing some a character, a comma, a bracket, a semi-colon. Rarely does the error message point you to the error but rather to something else affected by the error.
It would seem in this day and age with hex core processors running at multi-ghz clock rates that error processing could be improved. If you start an array statement, and have missing or wrong element separators, report that and highlight the location in the source file. If you have a structure that is not properly terminated, report that rather than the effect of that on another statement.

I realize Arduino is found on GCC so perhaps this should/could be addressed in the GCC but it does seem that error reporting is still stuck in the 70's, not much removed from punch cards and cryptic error numbers.

Just my two cents, or maybe 2 dollars with today's inflation.

Thanks,
Jim Brooks

rogerclarkmelbourne, 2015-04-07:
Hi Jim

The Arduino IDE uses standard open source compilers, specifically GCC.

The error messages they generate are IMHO the norm when programming in C or C++

Some newer languages may have better error reporting, as may some commercial C compilers, but the errors messages that gcc generates are pretty much par for the course as far as I'm concerned.

Btw. I too remember the 8 and 16 bit micros, like the 8080 and the 6502, which I used to program in assembler, and I have been using c compilers on and off for more years than I care to mention ;-)

jbhome, 2015-04-10:
Hi Roger 
I needed to send a stp file to a computer from my phone 
I located gallery picked the file and then share via email to send to the other computer
A painless exercise hiding the complexity of the task 
It is unfortunate with so many inovations that good enough is accepted In this case
Many of the ui improvements not only make a process more friendly they save time
Imagine the steps to send the file without the share function. So you think the gcc group should get the memo?

VaclavSal, 2015-04-10:
I have to concur with OP. I do not read Arduino web site 24 /7 , but a very brief scan of majority of " what the heck is this error telling me "  posts should also support the OP. 
And I do not buy it is GCC fault / standard etc.
Such "opinion", and you have every right to have it,  is the "dancing bear"  case - straight from A. Cooper's " Inmates are running the asylum". 
Every software "engenerd" should be required to read this book and maybe they will  enter 21st Century someday as a profession.