arduino/Arduino #2918: Due Core files do not show up in hardware (still show up in Boards menu)

Asaurus1, 2015-04-06:
I need to edit the variant.cpp file for the Due. In previous versions this was located in the hardware subdirectory. In 1.6.3 it seems that the Due core files are not even installed in the arduino-1.6.3 directory. The fact that these files are "missing" is a problem, even though I can still compile to the Due like before.

ffissore, 2015-04-07:
@Asaurus1 The IDE does no longer bundle SAM files. They need to be installed using boards manager. I've added an entry to Boards Manager FAQ: https://github.com/arduino/Arduino/wiki/Boards-Manager-FAQ#where-are-installed-cores-located

closed 2015-04-07 by ffissore

Asaurus1, 2015-04-07:
You might consider adding a "open core folder" link to the board manager entries.

ffissore, 2015-04-07:
Right, and you're not alone with this feature request #2874