arduino/Arduino #2919: [DOC] Inconsistency in Stream documentation and derived libraries (Serial, Ethernet, SD, Wire)

q2dg, 2015-04-06:
In http://arduino.cc/en/Reference/Stream says "Some of the libraries that rely on Stream include: Serial, Wire, Ethernet Client, Ethernet Server, SD". When I go to Reference page of these libraries -except Serial- I can see listed some of Stream methods, like .available(), print(), .read()...but not others like find/findUntil(), readBytes()/readBytesUntil(), readString()/readStringUntil(), parseInt()/parseFloat() and setTimeout()...

I think documentation should be refactored to list in one point common methods (Stream page) and then, in library reference pages a link to these (cleaning some redundancies). Maybe I'm wrong...but nowadays it doesn't clear if I can write EthernetServer.readString() or not, for instance.

per1234, 2015-05-02:
I agree. When I was learning to use the Ethernet library it was only by coming across an example sketch that I realized I could use the Stream functions. It was very confusing to see the usage of functions that were not shown in the Ethernet library documentation. Even though it says "read() inherits from the Stream utility class" at the top of the Ethernet library Client class read() documentation, that still means nothing to the beginner that Arduino is intended to be accessible to. In the EthernetUDP class read() documentation it is not even mentioned.

agdl, 2015-09-04:
@q2dg I'm marking this as duplicate and closing this because is a duplicate of https://github.com/arduino/Arduino/issues/3718

closed 2015-09-04 by agdl