arduino/Arduino #2925: Platform libraries are compiled for both AVR and SAM

sej7278, 2015-04-06:
In 1.6.3 the platform libraries are compiled for both AVR and SAM using avr-g++

```
avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR
-I~/arduino-1.6.3/hardware/arduino/avr/cores/arduino
-I~/arduino-1.6.3/hardware/arduino/avr/variants/standard
-I~/arduino-1.6.3/libraries/Servo/src
~/arduino-1.6.3/libraries/Servo/src/avr/Servo.cpp -o /tmp/build4733981319364822328.tmp/Servo/avr/Servo.cpp.o

avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR
-I~/arduino-1.6.3/hardware/arduino/avr/cores/arduino
-I~/arduino-1.6.3/hardware/arduino/avr/variants/standard
-I~/arduino-1.6.3/libraries/Servo/src
~/arduino-1.6.3/libraries/Servo/src/sam/Servo.cpp -o /tmp/build4733981319364822328.tmp/Servo/sam/Servo.cpp.o
```

Is that not wasting time, especially as the SAM object file wouldn't work on a SAM board anyway as its using avr-g++ not arm-none-eabi-g++

ffissore, 2015-04-07:
That's expected unfortunately. `arch` folder has been removed and sources are no longer selected using target arch: everything under `src` is compiled. `avr` and `sam` folders inside `Servo` are really just documentation, a hint for the human, not something the IDE uses for some kind of decision
/cc @cmaglie @facchinm

closed 2015-04-07 by ffissore

cmaglie, 2015-04-07:
There is a good amount of discussion around the library format and the `arch` folder:

[1] https://groups.google.com/a/arduino.cc/d/msg/developers/SbCg3lPphWI/C97fLB9V-WUJ
[2] https://groups.google.com/a/arduino.cc/d/msg/developers/ku32KnasSm0/HOWD4IDgeIUJ
[3] https://groups.google.com/a/arduino.cc/d/msg/developers/4Jk4P0rsWPc/twLFMdkMdMsJ
[4] https://groups.google.com/a/arduino.cc/d/msg/developers/m2XEF4-uFxI/b0BQDbY-q_8J

To make the long story short: there is no agreement about the `arch` folder and it's usefulness. The community seems really sensible about changing format of Arduino libraries, so the `arch` folder feature has been disabled.

> Is that not wasting time, especially as the SAM object file wouldn't work on a SAM board anyway as its using avr-g++ not arm-none-eabi-g++

Yes, in the case of the Servo library the IDE may save one call to `g++` if the arch folder feature was available (the whole SAM-specific code is disabled by `#ifdef` guards so it compiles fine).
On the other side it seems that the number of libraries that can actually get benefit from this kind of processing, without a deep refactoring, is very low. Maybe at some point we'll re-enable this feature but this will happen only if this feature is backed by a real and strong need from the community.