arduino/Arduino #2926: "malformed" version number in library.properties causes IDE to exit

bperrybap, 2015-04-07:
When the IDE starts up it appears to scan the library.properties files of each of the libraries under the users sketchbook directory.
If it encounters a version number that it doesn't like, it prints an error to stderr and exits.
Unfortunately, at least on linux, stderr messages are not seen when the user starts the IDE by clicking on the arduino startup script.

It seems inappropriate for the IDE to abort just because a library may have a version number that it does not like. In my view it could (should) ignore these strict rules (yeah, yeah, that would be against the semver spec); however, when there is not an update check or there is no way of doing an update, the contents of the version string are pretty irrelevant, so there is no need to be so strict.

At a minimum if the strict naming convention is to be enforced and the IDE is going to abort, it should at least put a dialog so that users can figure out what the problem is and then either abort or ignore that library.

However, My preference would be to relax the checking and use the version string "as is" whenever possible, which is probably most of the time.

wayoda, 2015-04-07:
I can confirm that the IDE doesn't start when a malformed library.version is encountered.
Here is a example for *version* "brutzi.1.1"
```
java.lang.NumberFormatException: For input string: "brutzi"
        at java.lang.NumberFormatException.forInputString(NumberFormatException.java:65)
        at java.lang.Integer.parseInt(Integer.java:492)
        at java.lang.Integer.valueOf(Integer.java:582)
        at cc.arduino.contributions.VersionHelper.valueOf(VersionHelper.java:14)
        at cc.arduino.contributions.packages.DownloadableContribution.getParsedVersion(DownloadableContribution.java:97)
        at cc.arduino.contributions.libraries.LibrariesIndexer.scanLibrary(LibrariesIndexer.java:160)
        at cc.arduino.contributions.libraries.LibrariesIndexer.scanInstalledLibraries(LibrariesIndexer.java:122)
        at cc.arduino.contributions.libraries.LibrariesIndexer.rescanLibraries(LibrariesIndexer.java:102)
        at cc.arduino.contributions.libraries.LibrariesIndexer.setLibrariesFolders(LibrariesIndexer.java:90)
        at processing.app.BaseNoGui.onBoardOrPortChange(BaseNoGui.java:787)
        at processing.app.Base.onBoardOrPortChange(Base.java:1126)
        at processing.app.Base$10.actionPerformed(Base.java:1283)
        at processing.app.Base.rebuildBoardsMenu(Base.java:1259)
        at processing.app.Base.<init>(Base.java:258)
        at processing.app.Base.guardedMain(Base.java:198)
        at processing.app.Base.main(Base.java:113)

```
The IDE should not exit here, but show an error dialog and disable the library.
Relaxing the version check is not an option because Its just so dead simple to play along the rules....

bperrybap, 2015-04-07:
Of course it is an option to relax the rule checking. It is only s/w after-all.
While it should be simple to play by the rules, there is an old adage with respect interoperability:
"Be very strict with what you send, but be very liberal with what you accept". 
If you take this to heart, it does wonders to not only help interoperability, but also can help reduce support loads, since many restrictions (like this one) are artificial limitations and removing the limitation can make something work that would have not worked otherwise.

One problem that I see with just showing a dialog box and disabling the library is that the library manager does not offer a way to remove the offending library. Seem silly to me, but that was a conscious decision. That means that the user can install a library with the IDE that breaks the IDE, but can't clean it up using the IDE and must use a file manager or shell to remove the library.

Maybe the library manage should refuse to install a library with an invalid version as well?
(I haven't checked to see if it already does this)

commit b9a90f69 (2015-04-07): Invalid versions don't cause IDE to crash and exit any more. They are reported and contributions are considered missing version. Fixes #2926
closed 2015-04-07 by ffissore