arduino/Arduino #2930: ARDUINO 1.6.3: Sketch uses X bytes (Y%) of program storage space. Maximum is 32,256 byte

dubnikov, 2015-04-07:
While compiling on ARDUINO 1.6.3 the sketch (attached) I got this error:
Sketch uses 49,304 bytes (152%) of program storage space. Maximum is 32,256 bytes.
Global variables use 176 bytes (8%) of dynamic memory, leaving 1,872 bytes for local variables. Maximum is 2,048 bytes.
processing.app.debug.RunnerException: Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
   at processing.app.debug.Compiler.size(Compiler.java:336)
   at processing.app.debug.Compiler.build(Compiler.java:119)
   at processing.app.Sketch.build(Sketch.java:1170)
   at processing.app.Sketch.build(Sketch.java:1143)
   at processing.app.Editor$BuildHandler.run(Editor.java:2016)
   at java.lang.Thread.run(Thread.java:745)
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.

Sketch:
http://www.instructables.com/files/orig/FLH/1QCH/I0C982BD/FLH1QCHI0C982BD.rar

But the same sketch on ARDUINO 1.0.6 compile without errors at all.

ffissore, 2015-04-07:
1.6.3 uses an updated gcc with an updated avrlibc. Updates usually mean more functionality that sometimes lead to bigger programs. While I don't know the root cause of your particular issue, if I was you I would: 1) stick with 1.0.6 for that particular sketch only OR 2) optimize it so that it fits into the board OR 3) play with gcc optimizations flags and try to squeeze (try switching from -Os to -O3)

PaulStoffregen, 2015-04-07:
I looked at your code briefly.

This seems to be a pretty terrible compiler bug.  It's trying to inline your huge move3Servos function. It's called in many places, which results in many copies that rapidly burn up all the code space.  Perhaps it's due to passing Servo objects as parameters (instead of passing them as references or pointers)?

You can get your code to compile with this.  I can't know if it really works...

        //////////////////////////// Servo functions ////////////////////////////////
        __attribute__((noinline)) 
        void move3Servos(Servo servo1Name, float target_pos1, Servo servo2Name, float target_pos2, Servo servo3Name, float target_pos3, float easing) {

dubnikov, 2015-04-07:
Terrible compiler bug but they closed the issue. :(

PaulStoffregen, 2015-04-07:
Well, I think so, but I can't speak for the official Arduino project.

My understanding is the compiler is supposed to use some heuristic algorithm to guess whether inlining a function is worthwhile.  Very small functions are supposed to be automatically inlined, because it's smaller and faster to just put their tiny amount of code into the calling function than suffer the overhead of the function call and return.

Why the new avr-gcc would try to inline your huge move3Servos function is a mystery to me.  But maybe it's some obscure issue with your use of passing the entire Servo object?  This is probably a question of for the avr-gcc devs.

ffissore, 2015-04-08:
@dubnikov I closed the issue because I don't have the smarts for hacking gcc :) Thanks to @PaulStoffregen for delving into the wilds of inlining.
Looking at gcc bugzilla, it looks like a similar issue has been [reported](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=38629) in the past and it's due to a too high `inline-call-cost` gcc param (set to 5 for AVR)

ffissore, 2015-04-08:
I've played a bit with gcc params to see if this can be somehow controlled from the outside. If I add `-fno-inline` the sketch compiles fine (same result as adding `__attribute__((noinline))`)
Should this param be added by default in platform.txt ?

ffissore, 2015-04-08:
/cc @cmaglie @facchinm

matthijskooijman, 2015-04-08:
@ffissore, Wouldn't that disable _all_ inlining? I think that would significantly hurt performance for all small functions that are correctly inlined now. At the very least, we should do a mass-compile of all kinds of examples for testing.

cmaglie, 2015-04-08:
This really looks like a compiler bug and from the previous comments from @ffissore and @PaulStoffregen looks like there are two easy workarounds too.

I'd like to not change the compilers defaults (that works for the 99.99% of the sketches until now) and eventually wait for a fix coming from the avr-gcc developers.

q2dg, 2015-04-08:
Will you inform avr-gcc developers about this issue, then?
Maybe reviving https://gcc.gnu.org/bugzilla/show_bug.cgi?id=38629 issue?

PaulStoffregen, 2015-04-08:
This same issue apparently was also being discussed on the forum.

https://github.com/arduino/Arduino/issues/2930

Bill Westfield and I put in duplicate effort to analyze this problem and separately discover the inline behavior, all because the original poster didn't have the courtesy to place links on both conversations to the other.

ffissore, 2015-04-09:
@dubnikov please revive [gcc issue on gcc bugzilla](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=38629) as @q2dg suggested. I would do that myself, but it implies attaching your sketch as a test case and I'm not authorized to play with your stuff. Once done, please report back here.

dubnikov, 2015-04-09:
@ffissore
Feel free to report, you can do it better than me.

Thank you in advance.

ffissore, 2015-04-09:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=38629#c7

WestfW, 2015-04-11:
Actually, I don't think that this is 38629.   It turns out that it is NOT avr specific; I can get the same bloating behavior (a bit worse, actually) editing the preprocessed file to yield x86 compatible code.
I've submitted <https://gcc.gnu.org/bugzilla/show_bug.cgi?id=65740>

~~~~
BillW-MacOSX-2<10423> /sw/lib/gcc4.7/bin/gcc-fsf-4.7 -c -Os -w foo.ii -fno-inline-small-functions
BillW-MacOSX-2<10424> size foo.o
__TEXT  __DATA  __OBJC  others  dec     hex
8725    209     0       0       8934    22e6

BillW-MacOSX-2<10426> /sw/lib/gcc4.7/bin/gcc-fsf-4.7 -c -Os -w foo.ii 
BillW-MacOSX-2<10427> size foo.o
__TEXT  __DATA  __OBJC  others  dec     hex
29936   209     0       0       30145   75c1
~~~~

q2dg, 2015-04-11:
Well, it seems there is already an answer...https://gcc.gnu.org/bugzilla/show_bug.cgi?id=65740#c1

cmaglie, 2015-07-03:
This turns out to be a gcc-bug, please follow up on the gcc's issue trackers: if there is someone able to fix the issue is there not here.

In the meantime, if your sketch belongs to the 0.01% that is affected by the problem, you can apply the workarounds suggested in this discussion.

closed 2015-07-03 by cmaglie

joelkap2, 2016-07-09:
Hi there, I have recently bought my first Aduino. I am struggling to even upload the blink example, this is killing me I have tried everything. I am needing to eventually understand how to use scratch in order to programme the Arduino but it would be great if I understood how to even upload an example... Could somone please help me with this issue.