arduino/Arduino #2931: Add to Library Manager: ArduinoThread

ivanseidel, 2015-04-08:
Link to repo: https://github.com/ivanseidel/ArduinoThread
Link to Release: https://github.com/ivanseidel/ArduinoThread/releases/tag/1.0

ffissore, 2015-04-08:
Added. However, your tag 1.0 already contains a library.properties with version 2.0. Therefore, the 2.0 version of published library is your tagged version 1.0. Please fix library.properties tagging a version=2.0.1 or something. FYI, we look at all tags of the git repo, not just the one you specified: this allow you to be independent from us when releasing a new version. Also see https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

ffissore, 2015-04-08:
Once you'll have tagged the fixed version, our job will publish it in at most an hour

closed 2015-04-08 by ffissore