arduino/Arduino #2936: SD library is GPL v3.0 not LGPL 2.1+  (copyright violation)

bperrybap, 2015-04-08:
The SD library that ships with the IDE always was and I assume still is licensed GPL v3.0 as that license is what is specified in the SD sources in Greiman's latest git hub repo.

The source files clearly state that the license is GPL 3.0 yet back in January 2015, a README.adoc file was added to the SD library that is claiming copyright of the SD library by Arduino LLC and is stating that the SD library is LGPL 2.1+

This seems to be a copyright violation as you cannot change source licensed as GPL 3.0 to  LGPL 2.1+

Unless you have a special license from William Greiman, and SparkFun to re-release it as LGPL 2.1+ , in which case all the source files would need to be changed to reflect the new license, the readme file in the SD library needs to be corrected to indicate the true copyright owners and that the SD library is licensed GPL v3.0 instead of LGPL 2.1+

ffissore, 2015-04-08:
I guess it's a copy paste mistake. Readme.adoc were all added in a shot. We'll change it asap

closed 2015-04-09 by ffissore

bperrybap, 2015-04-09:
Technically the updated copyright is incomplete.
SparkFun wrote a small wrapper layer on top of William Greiman's sdfatlib which is down in the utility directory.