arduino/Arduino #2937: OS X Buttons and Retina

Yaneeks, 2015-04-08:
Hey,

If u have a second, upgrade button design for Yosemite, becouse it's look just old.

ffissore, 2015-04-08:
Can you define "old" and "bad"? Some screenshots comparing the IDE to some other app may help us get the desired effect

BTW what IDE are you using? 1.6.3 brought a lot of UI improvements on macosx

Yaneeks, 2015-04-08:
http://monosnap.com/image/nhWZshaKt46h0P0d5p23Jpaq6Ej9bu
http://monosnap.com/image/ePSKY5irzxxx0w0VXKmsM5igii3nNd

I believe you can see this. In Yosemite every icon/select tab/input is flat and specific as you can see in comparision on the first screenshot.

That "old" font you can see on the second picture and i know this is symbolic font but it could looks better on Retina i think

Tell me guys what you thinking 'bout this :)

ffissore, 2015-04-09:
Current look and feel resembles Mavericks. Look and feel is managed by the Oracle JVM bundled with the IDE. I guess we'll have to wait for a update to have the latest yosemite them support.

closed 2015-06-19 by ffissore