arduino/Arduino #2944: Max78630 energyboard library add request

akkoyun, 2015-04-09:
hi all,
we developed a pollyphase measurement board an library 
https://github.com/x2bus/EnergyBoard
we want ta add library to arduino library manager  thanks

ffissore, 2015-04-09:
Hi. Your library is in Library folder. Our job currently does not support fetching subfolders. That may change in the future but for now it's a blocking issue: we cannot add your library to lib manager.

akkoyun, 2015-04-10:
i change the folder structure

ffissore, 2015-04-11:
Great! Now you just need to make a "release" of your library (add a tag) and it will become available in lib manager

akkoyun, 2015-04-12:
thats ok. (i think :) )

ffissore, 2015-04-12:
Still no tags: https://github.com/x2bus/EnergyBoard/tags

akkoyun, 2015-04-12:
thats ok now :)

ffissore, 2015-04-13:
Done. Thanks :)

akkoyun, 2015-04-21:
New version 1.1.0 relased what can i do ? 
is it included automaticly

cmaglie, 2015-04-21:
You must change the version in `library.properties` (it's the version in this file that counts not the tag)

ffissore, 2015-04-22:
Indeed. I've just made FAQ clearer https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-publish-a-new-release-once-my-library-is-in-the-list

akkoyun, 2015-04-22:
so ? i made some thing wrong ?

ffissore, 2015-04-23:
In your master, `library.properties` version is 1.1.0
In your tag 1.1.0, `library.properties` version is 1.0.2
In your tag 1.0.2, `library.properties` version is 1.0.1
In your tag 1.0.1, `library.properties` version is 1.0.1
Hence, we are able to publish version **1.0.1** from tag 1.0.1 and version **1.0.2** from tag 1.1.0 but I don't think that's what you really wanted.
To keep things simple, even if it's not needed, please use the same tag name as the version in your library.properties

akkoyun, 2015-04-23:
oke i fix tag 1.1.0 thanks i understand system thank u

ffissore, 2015-04-23:
Please make another tag, something like 1.2.0 with version 1.2.0 in library.properties

akkoyun, 2015-04-23:
done

closed 2015-04-23 by ffissore

akkoyun, 2015-05-16:
Hi,
our library name is changed from (max78630 - V1.2.0) to (EnergyBoard - from V2.0.0) 
but old library still at libmanager
can u delete old library (max78630)

thanks

akkoyun, 2015-07-25:
Hi,
our GitHub Repo removed to
https://github.com/LVXElektronik/EnergyBoard
link. can i do something for arduino library ?