arduino/Arduino #2949: Allow board manager to install third-party hardware add-ons

Lauszus, 2015-04-09:
It would be nice if the board manager allowed users to install third-party hardware add-ons. Either via the GUI or just CLI.

For instance it would allow them to install: https://github.com/Lauszus/Sanguino or https://github.com/TKJElectronics/Balanduino/tree/master/Firmware/hardware.

I know this was discussed on the mailing list as well, but I just want to create this issue to track it.

probonopd, 2015-04-09:
...or possibly even https://github.com/esp8266/Arduino

rogerclarkmelbourne, 2015-04-10:
I'm sure the Dev team are already aware of this. 

I've already mentioned this to them. I'm sure this is a duplicate issue, but I can't find the original issue at the moment

ffissore, 2015-04-30:
Fixed with #3055

closed 2015-04-30 by ffissore