arduino/Arduino #2952: Avrdude on Linux64 is compiled whitout libusb

eni23, 2015-04-09:
arduino: 1.6.3 release from arduino-website
os: debian wheezy amd64

the avrdude-binary is not compiled with libusb, which means no usb-serial adapters are supported.   since most arduino-boards depend on usb-communication this is a quite big impact. 
workaround: compile avrdude 6.0.1 on your own and replace broken binary's

mschlenker, 2015-04-10:
USB serial adapters usually are supported by the linux kernels drivers, not userland drivers. Those userland drivers are usually just needed for more exotic programmers. 

Could you please post the USB IDs of the USB serial adapter (lsusb output) and the kernel version you are using (uname -a). Then after plugging in the usb serial adapter take a look at the last lines in the out put of dmesg.

matthijskooijman, 2015-04-10:
Agreed that libusb isn't needed for usb-serial adapaters, but not having libusb is still a problem for non-serial programmers like the JTAGICE3 and some others.

ffissore, 2015-04-10:
I've checked the logs, and indeed since we switched to static compiling, avrdude is not linking to libusb anymore. Problem is without static, gcc fails with a missing LTO plugin.

ffissore, 2015-04-10:
This issue was moved to arduino/toolchain-avr#23

closed 2015-04-10 by ffissore

ffissore, 2015-04-23:
1.6.4 core and tools (with libusb fix) available through boards manager