arduino/Arduino #2954: Add to Library Manager: Pushetta

guglielmino, 2015-04-10:
Url to repo
https://github.com/guglielmino/arduino-pushetta-lib

ffissore, 2015-04-10:
You library is in a subfolder of the repo. We currently don't support this layout: libs have to be in the root. See also https://github.com/arduino/Arduino/issues/2944#issuecomment-91156647

ffissore, 2015-04-10:
Just added this: https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#is-my-git-repo-ok

ffissore, 2015-04-22:
Closing for lack of feedback. Reopen when lib is ready for lib manager

guglielmino, 2015-04-22:
sorry, I was quite busy and I have lost the feedback. Now library layout is right, how can I reopen the request. Thanks

ffissore, 2015-04-22:
Thanks. However you still need to tag it.

guglielmino, 2015-04-22:
I did it. I made a mistake in version, it's 1.0.0 in library.properties and 1.0 in tag, I hope this isn't a problem, eventually I make a new tag.

ffissore, 2015-04-22:
Lib added. Thanks

closed 2015-04-22 by ffissore