arduino/Arduino #2955: Mac os x 10.10 avrdude: stk500_recv(): programmer is not responding

aplodusken, 2015-04-10:
Hi! I've now tried everything, uninstalled and installed the FTDI driver at least 10 times and disabled and enabled the original driver the same amount of times. Followed a number of different solutions on various blogs but nothing is working.
Every time i try to upload a sketch I to either UNO or Nano boards via the USB interface I get the same message.

 Using Port                    : /dev/cu.usbmodem1421
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding

I've also tried different versions of the Arduino IDE 1.6.3 to 1.0.5.
It's the same thing on both my Macs. One is on Mac OS X 10.09.x the other on Mac OS X 10.10.x.
I can upload sketches if I us my ArduinoISP module but if I have a shield on I have to take that of and put it back on again. And then I can't use the serial port for debugging.
THIS IS SO FRUSTRATING!!!!! And it's doing my head in!

There seem to be a lot of people having the same problem, What can we do to fix this???
Is there a solution that works???
//Micke

PaulStoffregen, 2015-04-10:
The FTDI driver is completely unrelated to anything "cu.usbmodem".

You've posted your question in the wrong place.  This issue tracker is for bugs and issues in the Arduino software.  The forum is the place for help.  And in this case, my guess is you've erased the Uno's bootloader by using ISP programming, AND you're mistaking the errors for a "driver problem", when in fact there isn't any driver issue in play, AND the drivers you believe are at fault are completely unrelated to your hardware.

From an Arduino issues point of view, perhaps some sort of README or other file should be placed together with the FTDI drivers, with Windows, Mac and Linux specific info about how to identify boards where the FTDI is needed.

aplodusken, 2015-04-10:
Ok Sorry! I will post it on the forum instead, I just got frustrated I've been struggling with this for three days now. Thanks for your reply.

closed 2015-04-10 by ffissore