arduino/Arduino #2960: Auto Format suggestion for new feature - delete empty lines in editor

VaclavSal, 2015-04-10:
In my opinion (!) Auto Format is one of the most neglected  and poorly advertised (by site administrators)  and useful features of Arduino IDE.

It seems that it could gain more followers if 

empty , entire  white space empty  lines 

could be optionally also "auto" deleted.

Chris--A, 2015-04-11:
Maybe this would be useful if there are two or more empty lines, and all but one was removed: As removing every empty line may make things worse.

It would also be nice if it could fix the code of newbies who don't like many lines of code:

```arduino
void setup() {
  Serial.begin(9600);
  Serial.print("hi here is a number ");Serial.print(analogRead(A0),HEX);Serial.print(", and here is the compile time: ");Serial.print(__TIME__);
}

void loop() {

}
```

Should become

```arduino
void setup() {
  Serial.begin(9600);
  Serial.print("hi here is a number ");
  Serial.print(analogRead(A0),HEX);
  Serial.print(", and here is the compile time: ");
  Serial.print(__TIME__);
}

void loop() {

}
```

ffissore, 2015-04-11:
IDE uses [AStyle](http://astyle.sourceforge.net/) for formatting code. Its default conf file is [here](https://github.com/arduino/Arduino/blob/master/build/shared/lib/formatter.conf)
As mentioned in the same file, you can copy and customize it to fit your needs.
@Chris--A one of the default option is indeed `keep-one-line-statements`: removing it make auto format change your example sketch as you wrote it

@VaclavSal  another option is `delete-empty-lines` (which is not enabled by default).

Check them all out at http://astyle.sourceforge.net/astyle.html

VaclavSal, 2015-04-12:
Thanks, I guess next (silly) question could be - why is this not VISIBLY published?
But that is a different department to ask , right?  
Or standard answer is  "google it".

closed 2015-04-12 by VaclavSal