arduino/Arduino #2963: Serial errors at >= 230400 bps

Hexloader, 2015-04-11:
This is probably related to the 16u2 usb-serial firmware.

Problem: At 230.4 Kbps, sometimes the 16u2 echoes back what has just been written to it.

How to reproduce: consider this code:

```c
#include <avr/io.h>

#define BAUD_RATE 230400
#define SERIAL_2X_UBRRVAL(baud) ((((F_CPU / 8) + (baud / 2)) / (baud)) - 1)

char rx_buffer[256];
unsigned char rx_head;

void uart_putchar(char c) 
{
    while(!(UCSR0A & _BV(UDRE0)));
    UDR0 = c;
}

void uart_puts(char *s)
{
    while (*s) uart_putchar(*s++);
}

int uart_getchar()
{
    if (UCSR0A & _BV(RXC0))
        return UDR0;
    else
        return -1;
}

int main()
{
    int i, j, c;

    UBRR0 = SERIAL_2X_UBRRVAL(BAUD_RATE); // Set baud rate
    UCSR0A = _BV(U2X0); // double speed
    UCSR0C = (3 << UCSZ00); // 8,N,1
    UCSR0B = _BV(RXEN0) | _BV(TXEN0); // Enable receiver and transmitter
   
    for (i = 0; i < 1000; i++) {
        for (j = 0; j < 80; j++) {
            uart_putchar('.');
            if ((c = uart_getchar()) != -1) {
                rx_buffer[rx_head++] = c;
            }
        }
        uart_puts("\r\n");
    }

    uart_puts("While I printed dots, you typed: ");
    for (i = 0; i < rx_head; i++) {
        uart_putchar(rx_buffer[i]);
    }
    uart_puts("\r\n");
    return 0;
}
```

This will print 80000 dots while buffering up the incoming chars from the usart. At the end, it prints the buffered chars. If you send a bunch of 'x' from a terminal while the dots are being received, you get:

```
While I printed dots, you typed: xxxxxxxxxxxxxxxx.xxxx.xxxxxxxxxxxxxxxxxxxxxxxx.xx.xx.xxxxx
```

You should see just 'x', instead some dots are echoed back.

matthijskooijman, 2015-04-13:
Interesting. I wonder if there is some crosstalk / interference between the lines because of the high speeds, or if this is a software bug somewhere.

You use direct register manipulation in your example. Does that mean it doesn't occur when using the normal Arduino `Serial` object, or didn't you test that?

> If you send a bunch of 'x' from a terminal while the dots are being received,

Does that mean no dots are received when you do not type anything?

What hardware are you testing on? Uno r3?

Do you have a logic analyzer or scope to actually check what's being sent over the RX line?

Hexloader, 2015-04-13:
>  Arduino Serial object,

I haven't tried with `Serial`. I did try an interrupt driven implementation based on the AVR109 app note, same results as above.

> Does that mean no dots are received when you do not type anything?

Sending dots is just to force full-duplex operation. If I only RX *or* TX everything works fine. It's RXing while TXing (AVR standpoint) what doesn't work quite right. My test's purpose is to keep TX busy while I type some letters on a terminal. Most of the times UDR0 reads what I'm typing on the terminal (ie. 'x'), but sometimes it reads what I just wrote into it (ie. a dot).

> Uno r3?

Sorry, I missed mentioning this is on a Uno R3. I've got a Duemilanove (FTDI chip) with a fried AVR, I'll replace the fried AVR and repeat the test as soon as I can, if that helps.

> logic analyzer or scope

I do have a crappy pocket DSO pseudo-scope. Limited capabilities, but let me know if you want a specific capture done with it.

Hexloader, 2015-04-13:
Oops, I think I misunderstood this question:

> Does that mean no dots are received when you do not type anything?

That's right: if I don't type anything, I don't get any dots.

Also, when I get a dot, the actual letter I typed is lost. Ie. if I type `abcdefghi`, chances are that I get something like `ab.defghi`.

matthijskooijman, 2015-04-13:
> Sorry, I missed mentioning this is on a Uno R3. I've got a Duemilanove (FTDI chip) with a fried AVR, I'll replace the fried AVR and repeat the test as soon as I can, if that helps.

Yeah, that would help.

> I do have a crappy pocket DSO pseudo-scope. Limited capabilities, but let me know if you want a specific capture done with it.
 
I'm mostly interested if the RX pin actually shows dots (i.e. the 16u2 is sending the wrong stuff), or if it shows whatever you type (i.e. the messup happens on the 328 side).

mschlenker, 2015-04-13:
This is going to be funny. For my work on the port of the Yún bridge and supporting scripts to the Raspberry Pi I also ran into some trouble with baud rates 230400 and higher. In my case I also tested with @watterott RPi shield bridge with direct connection between the Raspberry and the 328. And I did use the standard Serial. 

In theory 1MBit/s or even two should work. With the ATmega32u4 on the Yún 0.5MBit/s are known to work quite reliable on the AVR core. Here are some tests:

http://arduino.stackexchange.com/questions/296/how-high-of-a-baud-rate-can-i-go-without-errors

Could we move the discussion away from the IDE? For me it looks more like some hardware issue. If needed I can also have access to some crappy osci.

Hexloader, 2015-04-13:
Tested with a Duemilanove, works fine at 230.4 Kbps:

`While I printed dots, you typed: zzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzz`

> I'm mostly interested if the RX pin actually shows dots (i.e. the 16u2 is sending the wrong stuff)

I will try to get that tomorrow.

matthijskooijman, 2015-04-14:
Hmm, so that suggests that the bug is either in the 16u2 / 16u2 software, or in the PCB of the Uno (or either of those triggers a bug in the 328 / 328 software, of course).

I'd think the 16u2 firmware is the most likely culprit here, but we'll see what your trace says.

@mschlenker, what do you mean "move away from the IDE"? Move the issue to some other github tracker? Or from Arduino entirely? I don't think this is just some interference effect that should be expected at these higher speeds, it seems there really is  some bug involved here...

mschlenker, 2015-04-14:
I just tested with Pro Mini Clone and the Arduino USB serial converter (16u2), dots were splattered in. So the 16u2s firmware seems to be the most likely candidate.

Hexloader, 2015-04-14:
Ok, the trigger in the DS203 is especially buggy but I eventually got this capture:

[image]

I was bashing the 'U' key (01010101b), which produces an easy to recognize square wave, then *that* showed up. Looked it up on ascii table... It's a dot.

The signal is otherwise clean. Constant 5v on RX while TX is busy, no electrical noise whatsoever.

mschlenker, 2015-04-15:
With an FTDI232:

```
mattias@barium:~$ hexdump /tmp/out.txt 
0000000 7878 7878 7878 7878 7878 7878 7878 7878
0000010 7878 7878 7878 7878 7878 7878 3f78 7878
0000020 7878 7878 7878 3f78 3f78 7878 783f 783f
0000030 7878 7878 3f78 7878 7878 7878 7878 7878
0000040 7878 7878 7878 7878 7878 7878 7878 7878
0000050 7878 7878 7878 783f 7878 7878 3f78 7878
0000060 7878 7878 7878 7878                    
0000068
```

0x3f = b00111111 while the dot is 00101110. What's going on here?  I tried to test with Raspberry Pis UART, but do not get reliable serial over 115200 on the RPi itself, even with 16MHz UART clock.

Hexloader, 2015-04-15:
> the osci shows, that it's the ATmega328 that behaves strangely

This is _before_ the data makes it to the 328p. Apparently, the 16u2 sent that.

> Which MCU did you put in the fried Duemilanove? 168 or 328.

328, but the Duemilanove (FT232) doesn't have the issue discussed here. 

> 0x3f = b00111111

I'm guessing those are RX errors that result in >127 ascii chars, which your terminal displays as '?'. I've seen RX errors in the duemilanove at 230.4 Kbps, but I blame it to the 3.5% difference in the FT232 and 328p baud rates. My issue is specifically about the 16u2 in the Uno R3 echoing stuff back.

mschlenker, 2015-04-15:
> I'm guessing those are RX errors that result in >127 ascii chars, which your terminal displays as '?'. ... My issue is specifically about the 16u2 in the Uno R3 echoing stuff back.

Well, but receive errors should be more random, not always the same sequence?

Hexloader, 2015-04-15:
> should be more random, not always the same sequence?

Not if we consider the 8N1 frame format: 0 (start) + LSB + ... + MSB + 1 (stop) and the baud rate difference. The closest we can get to 230.4K at 16 MHz is 222.2K, which means the 328 is sampling the line about 3.5% **slower**. By the time it gets to the MSB, there is a chance that we are actually sampling the stop bit (1), which results in some char >= 128, which your terminal probably represents as '?' (at least screen does so).

I wanted to point out that the 16u2 in the Uno R3 is also clocked at 16 MHz and thus has exactly the same baud rate as the 328, so clock errors are simply not a problem in the Uno.

q2dg, 2015-06-08:
Maybe there's a possible solution here: https://groups.google.com/a/arduino.cc/forum/#!topic/developers/-AQ_7Fs0lBc