arduino/Arduino #2964: String not changeable (only ARDUINO DUE)

MichaDi, 2015-04-11:
The following code compiled for an ARDUINO DUE leads to a different result than the code compiled for AVR-based ARDUINOS (IDE-Version: 1.6.3):

void setup() {
  Serial.begin(115200);
}
void loop() {
  printOnDisplay("This is a test");
  while(1);
}
void printOnDisplay(char *string)
{
  string[4] = 0;
  Serial.println(string);
  if (string[4]) Serial.println("String hasn't changed!");
  else Serial.println("String has changed!");
}

On an ARDUINO DUE the serial output is:
"This is a test"
"String hasn't changed!"

On AVR-based ARDUINOS the output is:
"This"
"String has changed!"

matthijskooijman, 2015-04-13:
What you are doing there (changing a string constant) is invalid and not allowed by the C++ specification. On AVR it works, because there is no hardware support to mark memory pages as read-only. On the Due, I assume there is some mechanism for this.

The real problem here would be that warnings are disabled. With warnings enabled, the compiler would warn you that you're passing a string constant into a `char *` argument (as opposed to a `const char*`).

So, I don't think there is an actual bug here, apart from maybe the warnings (which are being tracked elsewhere). Ok to close this issue?

MichaDi, 2015-04-13:
Thank you for your answer, you can close the issue!

closed 2015-04-13 by ffissore