arduino/Arduino #2965: Platform.rewrite.keys.txt doesn't work on keys that contain substitutions

rogerclarkmelbourne, 2015-04-12:
It appears that rewrite keys doesnt work with this
'''
build.gcc_ver=gcc-arm-none-eabi-4.8.3-2014q1
compiler.path={runtime.ide.path}/hardware/tools/{build.gcc_ver}/bin/
'''
I think this may be because compiler.path={runtime.ide.path}/hardware/tools/{build.gcc_ver}/bin/ is getting changed to 

'''
compiler.path={runtime.ide.path}/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/ 
'''
before the rewrite keys is being processed.

In my case I removed the substitution, as its only used in one place, so did not cause a serious problem to modify

ffissore, 2015-04-13:
Additional details at https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/39#issuecomment-91954541
I think the issue is that you put a platform.keys.rewrite file into your core but  the IDE uses its own file only
platform.keys.rewrite is intended the make the IDE work with cores that are someway *old*, without changing them
Therefore its existence is something cores shouldn't be aware of (although cores that know and that don't want to be affected can use `rewriting=disabled`)

ffissore, 2015-04-30:
Closing as worksforme

closed 2015-04-30 by ffissore