arduino/Arduino #2966: sam: Missing support for HardwareSerial configuration

udoklein, 2015-04-12:
With commit https://github.com/arduino/Arduino/commit/1fab8c85e6d1b545ef88cb99965e0234bf3c2f8b on Jan 16 2014 the AVR version of HardwareSerial introduced a new method `void HardwareSerial::begin(unsigned long baud, byte config)` The SAM version https://github.com/arduino/Arduino/blob/master/hardware/arduino/sam/cores/arduino/HardwareSerial.h does not offer this kind of feature. According to the documentation http://arduino.cc/en/Serial/Begin I would have assumed that the SAM version offers this as well. How can this be fixed? Reason why I would need it: in my DCF77 project I have a "Meinberg Emulator" http://blog.blinkenlight.net/experiments/dcf77/meinberg-emulator/. This Emulator requires serial communication with mode 7e2. I want to port this to support both AVR and SAM.

cmaglie, 2015-04-13:
Unfortunately, the 2-stop-bits modes are available only on the alternative serial ports of the Due (`Serial1`, `Serial2`, `Serial3`) but not on first one (`Serial`).

The reason is that pins from 14 to 19 (`Serial1/2/3`) have been mapped to SAM3X's USARTs that supports all the serial modes, instead pins 0/1 (`Serial`) on UART that support only a few modes.

Here the reference code:
https://github.com/arduino/Arduino/blob/master/hardware/arduino/sam/cores/arduino/UARTClass.h#L28
https://github.com/arduino/Arduino/blob/master/hardware/arduino/sam/cores/arduino/USARTClass.h#L28

udoklein, 2015-04-14:
Aha, got it. This implies that I can have 7e2 if I am willing to connect to pins other than 0, 1. Or i have to use the native USB port to emulate a serial port. Thanks for pointing me in the right direction.

closed 2015-04-14 by udoklein