arduino/Arduino #2968: [DOC] Missing references to boards capables of using Keyboard & Mouse libraries

q2dg, 2015-04-12:
In http://arduino.cc/en/Reference/HomePage appears both libraries can be used only in Leonardo and Due but Micro and Esplora aren't mentioned.
Also, in http://arduino.cc/en/Reference/MouseKeyboard, Esplora is missing again.

agdl, 2015-04-17:
In order to also include the Yùn and other future Leonardo based boards i corrected the statement in
Leonardo based boards

q2dg, 2015-04-17:
Ok, but then...there is any place where is clearly shown which boards are Leonardo-based?

q2dg, 2015-04-27:
@agdl ...

q2dg, 2015-04-27:
@ffissore Thanks

agdl, 2015-05-07:
@q2dg a leonardo based board is a board that uses a 32u4 micro-controller and i think is well known. Where dou you wanna see this statement?

q2dg, 2015-05-07:
Well, I think "32u4 based boards" is better, so if an user doesn't know if a particular board is 32u4 based or not, he/she could look at its "product" page.
I will close the issue
Thanks
PD: For consistency, I would do the change in http://arduino.cc/en/Reference/MouseKeyboard , too.

closed 2015-05-07 by q2dg

agdl, 2015-05-07:
modified thank you!