arduino/Arduino #2969: i2c example using reserved i2c address

bperrybap, 2015-04-12:
The example on this page:
http://arduino.cc/en/Tutorial/MasterWriter

Is using i2c address 0x4 which appears to be in the reserved address space.
0 to 7 are reserved.

Not sure what to pick for an address. It probabaly doesn't matter, but it would be best to update the example to avoid using an address from the reserved address range.

It's been a while since I've tested, but I believe that these reserved addresses don't work on the pic32 processors.

agdl, 2015-04-17:
@bperrybap you are right, but the example is intended to be used with two Arduino so the address soesn't matter. Thank you!

closed 2015-04-17 by agdl

bperrybap, 2015-04-17:
Yes it does!
Just because it doesn't matter on an AVR doesn't mean it doesn't matter on other chips.
The low addresses will not work on Pic32 chips. I have no idea what happens on some of the ARM based chips.
Why continue to promote an example that known to be incorrect and violates specifications?
Fix the code. It is a trivial fix.

matthijskooijman, 2015-04-17:
+1 for fixing this.

Nantonos, 2015-04-17:
Hello Arturo,

Friday, April 17, 2015, 11:07:04 AM, you wrote:

> @bperrybapyou are right, but the example is intended to be used
> with two Arduino so the address soesn't matter. Thank you!

That is a totally bogus answer.

And when someone builds on that example to talk between an Arduino and
something else?

Fix the example, please.

-- 
Best regards,
 Nantonos                            mailto:nantonos@epona.net

agdl, 2015-05-07:
I made a PR because the on-line page takes the code from git. I fix the problem in all the examples. Let me know if it is ok.

https://github.com/arduino/Arduino/pull/3100

agdl, 2015-05-07:
I adde this statement here:
www.arduino.cc/en/Reference/Wire

"However the addresses from 0 to 7 are not used because are reserved so the first address that can be used is 8. "

Is it ok in this way?

commit 5199b0f6 (2015-05-14): Fixes  #2969:

This fixes the Wire examples that uses I2C reserved address (from 0 to 7) substituting them with 8 that is the first one available and that can be used.

I also modified the wire reference
http://www.arduino.cc/en/reference/wire
according to this fact.