arduino/Arduino #2970: [DOC] Ethernet examples include wrong libraries??

q2dg, 2015-04-12:
I've seen all Ethernet examples (http://arduino.cc/en/Reference/Ethernet) only include SPI.h and Ethernet.h files, but when I add the Ethernet library from menu's list (IDE 1.6.3) several #include lines automatically  appear in my sketch (specifically Dhcp.h, Dns.h, EthernetUdp.h, EthernetServer.h, Ethernet.h, EthernetClient.h...but SPI doesn't appear...is SPI.h not required, then?):

I deduce Ethernet library has been split in several pieces...but I think this fact should be mentioned in documentation. Something like"If you only want to use EthernetServer" class, you only need include these files; if you don't want to use DNS functionality, you can remove that file, and so on. And, of course, code examples should be updated acordingly...

q2dg, 2015-04-12:
A similar issue happens with WiFi library.

And in a much more exaggerated form, with GSM library. Examples only include a "GSM.h" file, but IDE fills tens of #include lines!!?? Are they all necessary??

On the other hand, example shown in http://arduino.cc/en/Reference/Firmata seems to lack the line 
#include <Boards.h> ...

soundanalogous, 2015-05-17:
The documentation at http://www.arduino.cc/en/Reference/Firmata is a few years out of date. I maintain Firmata but don't have access to update the Firmata reference on the Arduino site.

Chris--A, 2015-05-17:
@q2dg 
Only the .h that is named the same as the library is required to import the library folder for compilation (done separately in a temp folder).

The Ethernet.h lib includes the other files, so no you do not need to include them also.
When adding a library, it automatically adds an include for each .h found. 

SPI.h is required in the sketch, as just like the Ethernet.h file, its name is used to find the actual library and copy it to the compilation folder. This feature is only run on .ino files, so the `#include <SPI.h>` is not seen in the Ethernet .h files.

I think the IDE team is working on fixing this '*feature*' by adding a recursive algorithm, which would also allow a lot more freedom with sketch file layouts.

q2dg, 2015-05-17:
Ok, thanks a lot, @Chris--A . We will wait to this new algorithm,then.
I hold open this issue if someone wants to add some more comments.

@soundanalogous , have you ask for having access to web doc? Maybe you'll get a response...or not

agdl, 2015-09-29:
@Chris--A if you look at the new Arduino builder https://github.com/arduino/arduino-builder developed by @ffissore we have a new algorithm that makes possible to include all the really needed libraries without include all of them in the sketch, so we can think to modify the examples in order to include only what is really needed

Chris--A, 2015-09-29:
Cool, I'll have a look. I can update the examples in my EEPROM lib if they no longer need the include also.In the next few days I'll have a nice update for it available for viewing.

ffissore, 2015-10-02:
Since 9fadb545da3a6ee630ce0040a0c5d9ffad353bec there is no need to manually add dependent libraries to a sketch. We'll update ethernet lib examples to that they will only include Ethernet and not SPI

closed 2015-10-02 by ffissore