arduino/Arduino #2971: Compiler "temp" files - Question / configuration or a  bug?

VaclavSal, 2015-04-13:
With about 30 h / cpp files being compiled into windows ...Temp directory I have noticed they DO NOT get erased / deleted and the directory keeps growing after each compile. 
Is this normal ? 

I am still happy  running 1.6.0 Nov issue.

ffissore, 2015-04-13:
It's a bug

mschlenker, 2015-04-13:
It's a feature. 

On Linux people are used to have /tmp erased upon boot time. I like being able to enable verbose compile messages, fish out the hex files out of /tmp after compiling and flash them to several AVRs - this is quite faster than compiling the one sketch over and over again...

Of couse it's a bug, but i suggest adding 

[_] Do not clear temporary files after upload

to the configuration interface to enable examining the compiled binaries or fishing out the hex file afterwards.

matthijskooijman, 2015-04-13:
The intention, IIRC, is that the temp files are deleted by the IDE when it quits. This prevents recompilation of files during a single session, doesn't clutter the temp directory, but also allows taking out the hex or elf file you just compiled. This seems like the proper way to handle this to me. If it doesn't work like that on Windows, it's a bug.

ffissore, 2015-04-13:
Indeed. [This code](https://github.com/arduino/Arduino/blob/master/arduino-core/src/processing/app/BaseNoGui.java#L136-136) tells the same story. However `deleteOnExit` is not recursive, hence the bug.

mschlenker, 2015-04-13:
Ah, I see, of course, since 1.5.x the intention is only to recompile changed files. The temporary directory is used for all sketches in one session. Thus it would also make sense to clean up upon start for the  case the Arduino IDE crashes or gets killed - and of course for all those people with several dozen MB old sketches lying in %TEMP%. Determining which buildNNNNNNNNNNNNN.tmp was created by Arduino should by quite easy by looking on the filenames found there.

mschlenker, 2015-04-13:
Please consolidate with 2724 https://github.com/arduino/Arduino/issues/2724

ffissore, 2015-04-15:
See #2980 for a proposed fix

PR #2980 (2015-04-15): Delete temporary files on exit
The IDE wanted but failed to delete temporary files on IDE exit. Feature restored.
Temp files will be preserved until the IDE is closed
Added a command line option `--preserve-temp-files` for preserving temp files when using the command line
Fixes #2971
/cc @VaclavSal @mschlenker @matthijskooijman
commit 3465fcf9 (2015-04-15): File.deleteOnExit is not recursive. Replaced by DeleteFilesOnShutdown shutdown hook. Fixes #2971
closed 2015-04-15 by ffissore