arduino/Arduino #2975: IDE 1.6.3: HardwareSerial Compile Errors on Assignment

mweber-ovt, 2015-04-13:
In 1.0.6, it is possible to make the following assignment (ATMEGA 2560)
HardwareSerial console;
console=Serial1;

In 1.6.3, this results in compile errors.

The issue is reproducible on the Mac and Windows versions of the IDE.

mschlenker, 2015-04-13:
...and is reproducible on Linux. Please try instead:

HardwareSerial console=Serial1;

Stricter compiler behaviour should not be filed as bug. Please close.

mweber-ovt, 2015-04-13:
mschlenker:  My problem is, it's no longer possible to switch ports so the same code either outputs to one serial port or another, as in the example below.  Unless I should be doing something different to accomplish the same purpose; let me know!

HardwareSerial console;

void setup(): {
initialize serial ports etc
}

void loop(): {
- do stuff
- depending on situation: either console2usb() or console2xbee()
console.print("message");
}

void console2usb() {
console=Serial;
}
.....
void console2xbee() {
console=Serial1;
}
.....

mschlenker, 2015-04-13:
Maybe like this? Please correct me if I am terribly wrong.

```c
HardwareSerial cons[2] = { Serial, Serial1 };
 
void setup() {
  // put your setup code here, to run once:
  cons[0].begin(9600);
  cons[1].begin(9600);
}

void loop() {
  // put your main code here, to run repeatedly:
  cons[0].print("Hello world!");
  cons[1].print("Hello anyone!");
}
```

matthijskooijman, 2015-04-13:
I actually think you should be using _references_ or _pointers_ here, instead of creating new instances and then copying over data. Even if this worked before, I don't think it was officially supported.

Looking at your example, you want to modify the `console` variable after it is initialized, inside a function, meaning you cannot use references (which can only be initialized once). So, use a pointer:

```
HardwareSerial *console = NULL;

void console2xbee {
  console = &Serial1;
}

void loop() {
  console->println("foo");
}
```

If you just need to call print/write on `console` (e.g. call begin on Serial/Serial1 directly), you can consider to instead use the more generic `Stream` superclass instead of `HardwareSerial`, which then allows writing to more kinds of streamy objects as well (in particular `SoftwareSerial`).

In any case, I don't think this is really a bug, so I'd like to close this issue?

mweber-ovt, 2015-04-13:
Thanks for the responses.  This is clearly an effect of the new HardwareSerial library, I can restore the original behavior by reverting to the version in 1.0.6, so I will close this issue now.

closed 2015-04-13 by mweber-ovt