arduino/Arduino #2977: Compiled Program freezes if Serial.begin is used directly after interrupt pin is attached.

matthiasmo, 2015-04-14:
My guess is that there are some problems with using only one interrupt. If delayMicroseconds is used in between it works.

My Code:<pre>
<span style="color: #00979C;">void</span> <span style="color: #5E6D03;">setup</span>(){
&nbsp;&nbsp;<span style="color: #D35400;">delay</span>(1000);
&nbsp;&nbsp;
&nbsp;&nbsp;<span style="color: #D35400;">pinMode</span>(beep, <span style="color: #00979C;">OUTPUT</span>);
&nbsp;&nbsp;<span style="color: #D35400;">pinMode</span>(enc1, <span style="color: #00979C;">INPUT</span>);
&nbsp;&nbsp;<span style="color: #D35400;">pinMode</span>(enc2, <span style="color: #00979C;">INPUT</span>);
&nbsp;&nbsp;<span style="color: #D35400;">pinMode</span>(enc3, <span style="color: #00979C;">INPUT</span>);

&nbsp;&nbsp;<span style="color: #D35400;">digitalWrite</span>(beep, <span style="color: #00979C;">LOW</span>);
&nbsp;&nbsp;
&nbsp;&nbsp;<span style="color: #D35400;">attachInterrupt</span>(enc1, rotaryEncoder, <span style="color: #00979C;">RISING</span>);
&nbsp;&nbsp;
&nbsp;&nbsp;&nbsp;<span style="color: #D35400;">delayMicroseconds</span>(50);
&nbsp;&nbsp;
&nbsp;&nbsp;<span style="color: #5E6D03;">for</span>(<span style="color: #00979C;">int</span> i=0; i&lt;16; i++){
&nbsp;&nbsp;&nbsp;&nbsp;<span style="color: #D35400;">pinMode</span>(solenoidsTemp[i], <span style="color: #00979C;">OUTPUT</span>);
&nbsp;&nbsp;&nbsp;&nbsp;<span style="color: #D35400;">digitalWrite</span>(solenoidsTemp[i], <span style="color: #00979C;">LOW</span>);
&nbsp;&nbsp;}
&nbsp;&nbsp;&nbsp;&nbsp;
&nbsp;&nbsp;&nbsp;<span style="color: #D35400;">delayMicroseconds</span>(50); 
&nbsp;&nbsp;&nbsp;&nbsp;
&nbsp;&nbsp;<span style="color: #D35400;"><b>Serial</b></span>.<span style="color: #D35400;">begin</span>(57600);
&nbsp;&nbsp;<span style="color: #D35400;"><b>Serial</b></span>.<span style="color: #D35400;">setTimeout</span>(100) ;
&nbsp;&nbsp;
&nbsp;&nbsp;beeping();
}

</pre>

matthijskooijman, 2015-04-14:
Weird behaviour. Some questions:
 - You say you're running Serial.begin() directly after attaching the interrupt, but there's also some pinMode / digitalWrite calls in between?
 - There's a bunch of other code in the sketch as well. Can you try removing as much code as possible, while making sure the freeze still occurs?
 - How can you tell the program freezes? Can you verify where it freezes (e.g. by toggling a led at various points in the program)?
 - What hardware and IDE version are you using?

matthiasmo, 2015-04-14:
Things I can answer right away:
IDE 1.6.3 running under Ubuntu 14.04.1 / Arduino Due
It freezes in attachInterrupt, if I swap attachInterrupt and Serial.begin it freezes at Serial.begin.
I tested it by setting the beep pin high.

I will test it more and report shortly.

matthiasmo, 2015-04-14:
I'm so sorry, I cant reproduce the problem now. I don't know what is the reason for that. Adding the delayMicroseconds fixed it for me last week.
There must be some other circumstances that trigger this.
Should I close the issue until I encounter it again to report more details?

matthijskooijman, 2015-04-14:
Yeah, that's probably best. You can just reopen this issue when you encounter it again. Thanks!

closed 2015-04-14 by matthijskooijman