arduino/Arduino #2978: 1.63  Errors font shade makes errors unreadable

jrd210, 2015-04-14:
i.62 has clear visible standout errors but on 1.63 I cannot read them at all on my MacBook and have to copy and print! Unusable for error checking at present so staying with 1.62

ffissore, 2015-04-14:
Can you please add some more detail? What's wrong with the errors: color, size, antialias? /cc @00alis

jrd210, 2015-04-14:
I think the color is the same as say 1.62 but it does not stand out 
against the black background and therefore is really hard to see, size 
is fine. This is on the MacBook Pro 27" screen if that helps. I didn't 
get a chance to try it on another computer or screen. John Dale

mschlenker, 2015-04-14:
Looks like #2901 https://github.com/arduino/Arduino/issues/2901

Paul and Mattias painting bike sheds. @jrd210 would you like to tinker with lib/theme/theme.txt to make suggestions on colors and fonts?

jrd210, 2015-04-14:
@mschlenker --thanks for suggestions, I will see which one works best, I will try the lib/theme/theme.txt first out of curiosity

jrd210, 2015-04-14:
Thx, see comments, will try.

jrd210, 2015-04-14:
Changed hex code for color--many thanks. put #669966 Green in and it 
works pretty good

closed 2015-04-23 by ffissore