arduino/Arduino #2981: Arduino boards and libraries should be installable without internet

WestfW, 2015-04-15:
In various situations, ranging from restrictive or expensive venues, to remote areas with poor internet service, it is desirable to be able to install the Arduino IDE (multiple compilers, and libraries) without a viable internet connection.

Ideally, I think you should be able to read the .json files from a directory, and have the content picked from the same place.  Perhaps as a preference: "install from: [Internet | <browse for location>"
(not unlike searching for new windows drivers, in fact.)

al1-24, 2015-06-06:
Another improvement would be to choose the directory where downloade files are installed, instead of C:\Users\<username>\AppData\Roaming\Arduino15.

per1234, 2017-07-03:
>Another improvement would be to choose the directory where downloade files are installed, instead of C:\Users\AppData\Roaming\Arduino15.

@al1-24 that's off topic for this issue but there is another issue for this feature request already: https://github.com/arduino/Arduino/issues/2874

While I'm commenting I would like to express my support for WestfW's request. This problem comes up fairly regularly on the Arduino forum and it's really difficult to coach a beginner through manual installation of a complex hardware package that is only intended for installation via Boards Manager. I remember a thread where I helped a user do this with Intel Curie boards ended up being around 20 pages long before they were successful.

aboshaddad, 2020-05-30:
I did something should be work to install board ZIP file offline and manually:
[I did something should be work to install board ZIP file offline and manually:
https://deve-sp.blogspot.com/2020/05/zip-boards-manager-how-to-install.html](https://deve-sp.blogspot.com/2020/05/zip-boards-manager-how-to-install.html)

JoshuaOlu, 2025-02-21:
Has this issue been officially fixed?