arduino/Arduino #2983: The kalmanfilter library is 2 times available in the library manager

jantje, 2015-04-15:
Following Library is 2 times available in the library manager in Arduino IDE 1.6.3 

https://github.com/TKJElectronics/KalmanFilter

ffissore, 2015-04-15:
Reported upstream

ffissore, 2015-04-15:
Thanks for the heads up!

Lauszus, 2015-04-15:
@ffissore I believe this happened as I changed all words in the name with capital letters: https://github.com/TKJElectronics/KalmanFilter/commit/f214b76246e996e5a579b5cb4b17fbcc3ae53028.

ffissore, 2015-04-16:
The duplicate needs to be manually removed

ffissore, 2015-04-23:
Done

Lauszus, 2015-04-24:
@ffissore it seems like there still is a duplicate? The one name "Kalman filter library" (version 1.0.0) should be removed.

ffissore, 2015-04-24:
First you need to remove tag 1.0.0

Lauszus, 2015-04-24:
Done.

ffissore, 2015-04-27:
Done

Lauszus, 2015-05-29:
@ffissore this problem is still present.

ffissore, 2015-06-03:
@Lauszus did you remove the tags with the wrong names? I'm setting up a weekly job that will regenerate the list completely, but if the offending tag is still there, the problem will keep on showing up

ffissore, 2015-06-03:
FYI I've added an entry in the FAQ https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#sorry-i-did-something-wrong-how-can-i-change-or-unpublish-an-already-published-library

Lauszus, 2015-06-03:
@ffissore yes there is only one tag: https://github.com/TKJElectronics/KalmanFilter/tags.

That might also solve the problem that the [USB Host Shield library](https://github.com/felis/USB_Host_Shield_2.0) is still at 1.0.0 despite the fact that I released 1.1.0 a while ago: https://github.com/felis/USB_Host_Shield_2.0/tags.

ffissore, 2015-06-04:
List re-re-generated. Close issue if ok

Lauszus, 2015-06-04:
Thanks that fixed the issue. Note that I can't close the issue, as it was opened by @jantje.

jantje, 2015-06-04:
I can :-)

closed 2015-06-04 by jantje