arduino/Arduino #2988: if statement only one line?

minoseah629, 2015-04-16:
Yesterday, I was using the IDE for a project I was doing.  I had noticed that when i used If else statements, I had to include { } even though my code met the condition was a 1 line of code. 

With my experience with C#, I am allowed to exclude the use of { } if it was only 1 line. 

Just a minor annoyance.

matthijskooijman, 2015-04-16:
The code is just compiled by a C++ compiler (gcc), so single-line ifs should be allowed.

Can you post a small code example and the corresponding compiler error to illustrate your remark?

minoseah629, 2015-04-16:
i looked at it again. i guess i had a different compiler error to cause me to assume it didnt work.  nevermind.

closed 2015-04-16 by minoseah629

matthijskooijman, 2015-04-16:
Thanks for reporting back, good to hear your issue is solved :-)