arduino/Arduino #2989: "ld returned 5 exit status" on Win XP

oldmaker, 2015-04-17:
Two sketches of mine (only those two), compiling fine with IDE 1.0.x, cannot compile with new IDE (on Windows XP): after the syntax check, appears a Windows error relative to ld.exe; then compilation stops with message "collect2.exe: error: ld returned 5 exit status".
Issue context:
IDE 1.6.1  or  1.6.2  or  1.6.3
O.S. Windows XP

The same sketches, with IDE 1.6.x, on another PC with Windows 8.1, compiles OK.
The same sketches, with IDE 1.0.x, compiles OK.

I found a (ridiculous) work-around: adding some totally useless global variables, the two sketches compiles OK also with IDE 1.6.x on Win XP:
`int y = 0;`
`int z = 0;`
`int w = 0;`
The first sketch needs at least 3 additional global variables, the second at least 6.
It doesn't matter the name, the type or the values; the only requirement seems to be the number of variables.

Similar issues on this site: #2768 and #2813

Verbose debug output of IDE 1.6.3, on Windows XP:

Loading configuration...
Initializing packages...
Preparing boards...
Starting...

Uso la libreria Wire nella cartella: C:\Programmi\Arduino\hardware\arduino\avr\l
ibraries\Wire

C:\Programmi\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -w -fno-exceptions
 -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32
8p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\P
rogrammi\Arduino\hardware\arduino\avr\cores\arduino -IC:\Programmi\Arduino\hardw
are\arduino\avr\variants\standard -IC:\Programmi\Arduino\hardware\arduino\avr\li
braries\Wire C:\DOCUME~1\Pasquale\IMPOST~1\Temp\build9152852205119054589.tmp\Dem
o_NHD0420CW-Ax3_I2C.cpp -o C:\DOCUME~1\Pasquale\IMPOST~1\Temp\build9152852205119
054589.tmp\Demo_NHD0420CW-Ax3_I2C.cpp.o
C:\Programmi\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -w -fno-exceptions
 -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32
8p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\P
rogrammi\Arduino\hardware\arduino\avr\cores\arduino -IC:\Programmi\Arduino\hardw
are\arduino\avr\variants\standard -IC:\Programmi\Arduino\hardware\arduino\avr\li
braries\Wire -IC:\Programmi\Arduino\hardware\arduino\avr\libraries\Wire\utility
C:\Programmi\Arduino\hardware\arduino\avr\libraries\Wire\Wire.cpp -o C:\DOCUME~1
\Pasquale\IMPOST~1\Temp\build9152852205119054589.tmp\Wire\Wire.cpp.o
C:\Programmi\Arduino\hardware\tools\avr/bin/avr-gcc -c -g -Os -w -ffunction-sect
ions -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DA
RDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Programmi\Arduino\hardware\arduino\avr\co
res\arduino -IC:\Programmi\Arduino\hardware\arduino\avr\variants\standard -IC:\P
rogrammi\Arduino\hardware\arduino\avr\libraries\Wire -IC:\Programmi\Arduino\hard
ware\arduino\avr\libraries\Wire\utility C:\Programmi\Arduino\hardware\arduino\av
r\libraries\Wire\utility\twi.c -o C:\DOCUME~1\Pasquale\IMPOST~1\Temp\build915285
2205119054589.tmp\Wire\utility\twi.c.o
C:\Programmi\Arduino\hardware\tools\avr/bin/avr-gcc -c -g -Os -w -ffunction-sect
ions -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DA
RDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Programmi\Arduino\hardware\arduino\avr\co
res\arduino -IC:\Programmi\Arduino\hardware\arduino\avr\variants\standard C:\Pro
grammi\Arduino\hardware\arduino\avr\cores\arduino\hooks.c -o C:\DOCUME~1\Pasqual
e\IMPOST~1\Temp\build9152852205119054589.tmp\hooks.c.o
C:\Programmi\Arduino\hardware\tools\avr/bin/avr-gcc -c -g -Os -w -ffunction-sect
ions -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DA
RDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Programmi\Arduino\hardware\arduino\avr\co
res\arduino -IC:\Programmi\Arduino\hardware\arduino\avr\variants\standard C:\Pro
grammi\Arduino\hardware\arduino\avr\cores\arduino\WInterrupts.c -o C:\DOCUME~1\P
asquale\IMPOST~1\Temp\build9152852205119054589.tmp\WInterrupts.c.o
C:\Programmi\Arduino\hardware\tools\avr/bin/avr-gcc -c -g -Os -w -ffunction-sect
ions -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DA
RDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Programmi\Arduino\hardware\arduino\avr\co
res\arduino -IC:\Programmi\Arduino\hardware\arduino\avr\variants\standard C:\Pro
grammi\Arduino\hardware\arduino\avr\cores\arduino\wiring.c -o C:\DOCUME~1\Pasqua
le\IMPOST~1\Temp\build9152852205119054589.tmp\wiring.c.o
C:\Programmi\Arduino\hardware\tools\avr/bin/avr-gcc -c -g -Os -w -ffunction-sect
[... 390 lines cut ...]
   command(0xF1);     //  phase2=15, phase1=1 (default: 0x78)
   command(0xDB);     // set VCOMH deselect level:
   command(0x40);     //  VCOMH deselect level=1 x Vcc (default: 0x20=0,77 x Vcc)
   command(0x78);     // OLED characterization: OLED command set disabled (SD=0) (exit from OLED command set)
   command(0x28);     // Function set: fundamental command set (RE=0) (exit from extended command set)
   command(0x01);     // Clear display
   delay(5);          // Waits 5 ms (after a clear display, a minimum pause of 1-2 ms is required)
   command(0x80);     // Set DDRAM address 0x00 in address counter (cursor home) (default value)
   command(0x0C);     // Display ON/OFF control: display ON, cursor off, blink off
   delay(300);        // Waits 300 ms for stabilization purpose after display on
}
// _______________________________________________________________________________________

void loop(void)                       // MAIN PROGRAM
{  
      output();                       // Execute subroutine "output"
      delay(2000);                    // Waits 2 sec (only for visual effect purpose)
      blocks();                       // Execute subroutine "blocks"
      delay(2000);                    // Waits 2 sec (only for visual effect purpose)
}

Windows crash messages:
[image]
[image]
[image]

Coding-Badly, 2015-04-20:
Do you have Cygwin installed?

Coding-Badly, 2015-04-20:
Have you tried updating the Microsoft C Run Time?

oldmaker, 2015-04-21:
1) No, I don't use Cygwin.

2) My system is up-to-date (as for XP); I see that is installed Microsoft Visual C++ 2010 x86 Redistributable ver. 10.0.40219; checking on Microsoft site and on Windows Update, it seems to be the last version.

I also tried disabling every possible software, without success.
It's very strange that adding some useless variables the problem disappears ....

oldmaker, 2015-04-23:
User f41_ardu discovered another workaround for this issue: replace original file ld.exe with the file of the same name copied from the previous IDE 1.0.6.
For details see this thread on Arduino forum: [url=http://forum.arduino.cc/index.php?topic=316146.0]1.6.3 ld returmed 5 exit status error with a struct in a class? [/url]

positron96, 2015-10-14:
I confirm this on 1.6.5. 
Problem appears regardless of program size, even the most simple programs fail to compile.

Had to downgrade to 1.6.0, seems no problem (but library is missing several useful functions, e.g EEPROM.update).

fabelec95, 2015-10-18:
I could suggest to create a Temp directory on C drive and then change TMP and TEMP system variable definition to C:\Temp.

Linkers are generally sensitive on paths content, so it is recommanded to set simpler and shorter system paths, specially on older operating systems.

1.6.5 do not fails any more on linking after this change on Windows XP.

positron96, 2015-10-18:
Is the issue with ld known already? What exactly does error 5 mean?

fabelec95, 2015-10-18:
I suggest this solution because it solves the problem on my side for my Arduino projects and those of my friends under Windows XP.
For true explanation, Arduino team have to investigate around linker error messages.

positron96, 2015-10-20:
Thank you! I will try this when I encounter this bug again

NicoHood, 2015-10-20:
Are you living in china or why the hell are you using XP?
I guess they wont fix XP issues if they dont affect any other (supported) windows.

positron96, 2015-10-20:
It's an educational facility that does not have resources to upgrade a dozens of computers to be able to run win7. It happens.

positron96, 2015-10-20:
Actually, did anyone post this bug upstream? It seems to be a gcc/mingw/avr-gcc bug (sorry, don't know what project is directly related)

cmaglie, 2015-10-21:
Googling for "ld returned 5 exit status" it seems that Arduino is not the only one that is suffering this problem. BTW I only found people complaining about the bug but nobody actually investigated the issue.

If someone has some insight about the real **cause** of the bug it would be really helpful.

positron96, 2015-10-31:
If anyone can read a procmon dump here, I can provide one.
Also, changing $TEMP to something without spaces (D:\Temp) did not help me. Still getting the same error.

positron96, 2015-10-31:
I've peeked in the ld sources and did not find any ````exit(5)```` or anything like error code 5. In Windows, however, error code 5 usually means "Access denied" and stack analysis seems to support it. When ld starts to crash, a ````ntoskrnl.exe!KiTrap0E```` is fired which in turn calls ````ntoskrnl.exe!MmAccessFault````. Googling it shows that it is called when Page Fault error occurs. This is the most I can squeeze out of procmon dump.

73-de-LZ, 2015-11-27:
The nasty error code 5 beat me today, find a cure by adding Serial.prinln(); at the end of void loop(). Nasty error - nasty solution, but i could not leave the things that way, so I spent few hours in investigation of forum. Test the old version of ld.exe from IDE 1.05, OK, work, but didn't like it, then added some global variables, also work fine, but this is not a solution, it is just a nasty trick. Then i try to do what fabelec95 said in the comment here and it work just fine. 
1. Make short name directory without spaces in C:, for example C:/TEMP, 
2. From the Arduino IDE 1.6.6 open "File"=>"Prefernces" and hit the link to the file "preferences.txt", close the IDE as it is written "(edit only when Arduino is not running)", so exit from all opened scetches and close Arduino IDE.
3. Open the file preferences.txt and add at the first row this line :
build.path=C:\TEMP
4. Save it and start Arduino again.
Good luck!

rva1945, 2016-02-09:
This is really weird. My sketch didnt' compile because of this sentence:

velocidadTren = velocidadTren-int(velocidadMaxima/aceleracionPasos);

all of them integer variables and <> zero.

I tried everything, from creating a float variable to store the calculation, to adding spaces between the variables and operators, to no avail.

But yes, there is a fix, weird too: I added this one in the declarations section:

char myvar ='hello';

and, believe or not, it fixed it! Now it compiles. I truly hope the Arduino boys are reading this.

mralexgray, 2016-04-03:
this _really_ hasn't been fixed yet.  **sad**.

cmaglie, 2016-04-04:
> this really hasn't been fixed yet? sad.

@mralexgray are you affected by this bug?

The cause of this bug is still unknown and I'm still unable to reproduce it on my (two) Windows XP boxes. All the analysis made until now makes me think about some memory corruption that happens inside "ld" and under some very particular conditions (that vary from PC to PC).
There are various workarounds discovered by users, just to name few of them:

- adding "fake" global variables
- changing the lenght of the strings in progmem
- even renaming the windows temp folder!

also there isn't a "common" workaround that works for everyone. What works for one user may not work for another. Moreover these are... well... workarounds and they don't give any clue on where the real bug is.

So unless someone takes the time to dig into the "ld" source and debug it, I guess that this issue will last for a long time...

walthays, 2016-04-07:
I run a classroom lab with 33 Windows XP computers and do a month long Arduino unit. This year was a disaster as nearly every computer ran into this problem. I worked around it each time by adding code (a useless for loop, a String variable that does nothing, a double variable that does nothing, etc.) I am attaching two files which fail to compile on 1.6.5 and 1.6.7. The good news is that they did compile on the "nightly build" from two days ago (1.6.9 s/t). I never had problems in previous years (I update the Arduino software in August before school starts on the image I use to clone all the PCs.) It would be great if this problem could be fixed. My plan if there is no fix is to downgrade to a much older version of the IDE. I'm sorry I don't know the version that last worked perfectly, but it would be whatever version was the released version mid-August 2014. (I renamed both uploaded files to .txt but they are .ino sketches.)

[compileError2.txt](https://github.com/arduino/Arduino/files/208678/compileError2.txt)
[compileError1.txt](https://github.com/arduino/Arduino/files/208679/compileError1.txt)

RDiazPR, 2016-04-11:
Adding six variables worked for me. (XP, IDE 1.6.7) . I'm ok with this strange workaround for this strange bug...

Thanks @oldmaker !

mastrolinux, 2016-04-11:
@walthays do you want to became an Arduino Create Editor betatester? We are interested in supporting legacy systems and we can probably help you running the latest Arduino Editor in your lab. Please contact me at webmaster at arduino.cc

GoForSmoke, 2016-05-21:
Following  73-de-LZ commented on Nov 27, 2015 I did this and it works.
Why I think it works is simple. Somewhere the long pathnames overfilled a line buffer that now is not.
Maybe Murphy will prove me wrong but for now I can proceed.

1. Make short name directory without spaces in C:, for example C:/TEMP,
2. From the Arduino IDE 1.6.6 open "File"=>"Prefernces" and hit the link to the file "preferences.txt", close the IDE as it is written "(edit only when Arduino is not running)", so exit from all opened scetches and close Arduino IDE.
3. Open the file preferences.txt and add at the first row this line :
build.path=C:\TEMP
4. Save it and start Arduino again.

alainsamoun, 2016-06-22:
Worked for me with 6 variables added and IDE 1.6.9:
SIM800L on Win XP

HiTechHiTouch, 2016-08-12:
May I humbly suggest that this is a gnu compiler set gcc related problem?  

LD is a gcc program that links together the various separately code pieces (which I will call object modules).  The format of these input pieces are all under control of gcc, and they are all produced by gcc compilers.  The error in LD is a function of what the gcc tools (compiler) produced as output and fed into LD for further processing.

In other words, we have valid c++ sketchs that gcc won't compile properly.  Valid c++ (and invalid c++ for that matter) should not blow away the compiler toolset.  The problem is in the tool set, not the editor which prepared the input to gcc.

1.  All code related circumventions are intended to influence the compiler output by changing the content of the object module, and indirectly, its format.  For example, adding more variables may cause a "code section" in the object module to be slip or extended, and perhaps cause new or additional table entries documenting the code section and attributes.  

(I'm not an expert on gcc object modules, but I have a CS degree and worked for the department in a large computer company which maintained the LD equivalent.)

**Now let us assume a "bug" in LD, triggered by specific conditions in the content of object modules.**

Tweak the object module input so that it no longer meets the specific condition, and you have circumvented the problem.  And the only way an Arduino sketch writer has to tweak the object module is to tweak the code going into the compiler by adding/removing/reordering variables (classes, storage properties, etc).

2.  The circumvention of changing versions of LD should be obvious.  A different version may process object modules differently, have different internal table structures, etc. so that the bug "moves around" between different LD/object module combinations.  The game is to find a combination of "your object module"/"some version of LD" that works for you.

3.  The fact that the problem seems to predominate on XP systems MAY mean that LD is doing something with memory management that is illegal or triggers an XP memory bug.  Such happenings are rare, but not unknown.  

This could be DEP or something, a part of Windows that has been evolving since released the first version of the feature.  

It could be as simple as not checking for a specific return code from a memory management call.  (Reading Raymond Chen's "Old New Thing" blog gives instances where an interface was tweaked in a following release of windows because of how it was being used (incorrectly) by some applications.)

But in any case, a well behaved application isn't expected to see these problems.  It's easy to argue that if the large majority of applications can do it right, then so can GCC.  Application programmers who encounter windows problems recognize that they have to code around them _today_,  even if they eventually get fixed.

**The bottom line is that this problem belongs in the GCC court, not the Arduino IDE court.**

They should package up a "temp" directory for a trivial example (and we have to submit one!), then open a gcc problem with the example.  Arduino can't debug gcc, and dollars to doughnuts this is gcc.

elektor-labs, 2016-08-31:
I am seeing this exact problem (ld returned 5 etc)  since I started using 1.6.10 & 1.6.11. Sometimes compilation works fine, sometimes even compiling an empty sketch produces this error. (Windows 7 family, SP1) Never, ever had this problem before with any other version. Compiling the same sketch with 1.6.7 works fine. Since 1.6.7 the IDE seems to be getting worse with every new release. Since 1.6.10 the IDE uses avr-gcc 4.9.2, so the problem may very well be there.

HiTechHiTouch, 2016-09-01:
I started the Arduino IDE under Windbg, hoping for a first chance exception when some memory access violation or system check occurred.  I eventually got the LD code 5 (actually 0xC0000005) but the debugger never trapped.

I'm not a online Windbg expert (I've only used it for dump post mortems) so I can't say I had it set properly.  Still, it was a great disappointment, but I won't give up.

BTW, I got myself into the LD error 5 condition, restarted the IDE and still had the LD error 5.  I then restarted the IDE from the debugger, and you guessed it, the LD error 5 went away.  Continuing to run with the debugger for a few more hours eventually got another LD error 5 condition, but Windgb failed to notice.

positron96, 2016-09-02:
@HiTechHiTouch you need to debug ld.exe where the error started, not he IDE. I'm not Windbg expert but it probably tracks only exception in target process, and ld.exe is a separate one.

HiTechHiTouch, 2016-09-03:
Research turned up [How to debug a process as soon as it starts with WinDbg or Visual Studio 2010 ](https://bugslasher.net/2011/03/26/how-to-debug-a-process-as-soon-as-it-starts-with-windbg-or-visual-studio-2010/) which should be very helpful.  More news later.

cmaglie, 2016-09-05:
@HiTechHiTouch 
thanks for looking deeply into this issue, BTW one thing that makes this issue harder to fix IMHO is the lack of a reproducible example (that we could possibly pass to avr-gcc developers to look into).

I've tried all the examples posted here (and in many forums threads), but I never managed to reproduce this bug. This means that having just the sketch is not sufficient to reproduce it, it depends on other factors (paths? service pack installed? something else? who knows...).

One thing that we never tried is to zip all the build artifacts (`*.o`, `*.a`, etc..) produced by the Arduino IDE just before the `ld` command is executed. Maybe this would help to reproduce the bug? Can you zip the build folder and publish it somewhere?

PaulStoffregen, 2016-09-05:
In a couple workshops I recently hosted, we saw some machines fail to compile known good code.  The 2 cases I recall were a netbook and a mac that had an incredible number of other programs running.  On the mac, simply quitting some of the other programs solved the problem.  Nothing seemed to help the netbook.

Newer versions of Arduino use updated to newer versions Java runtime.  They are tasked with managing much more data using Java's hashmaps.  On my Linux 64 bit system with Arduino 1.6.11 running with 3 windows open, "top" says the java has allocated an astounding 12.7 gigabytes of virtual memory, with 0.73 gigabytes actually resident/used:

      PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND     
    25801 paul      20   0 12.758g 731612  35900 S   1.3  2.2   2:10.09 java

Now the builder program, which is growing in size & complexity, is also resident in memory.  All that, plus other programs, takes away virtual memory before the compiler can run.

Perhaps newer versions of the linker are wanting more virtual memory, which simply isn't always available, especially on these older computers still running Windows XP?

cmaglie, 2016-09-05:
> Perhaps newer versions of the linker are wanting more virtual memory, which simply isn't always available, especially on these older computers still running Windows XP?

This won't explain why adding some variables at the beginning of the sketch, or changing TEMP folders, solves the problem for some people.

elektor-labs, 2016-09-05:
For me it happened on Windows 7, not XP.

The first time I used 1.6.10 I encountered this problem but did not investigate and switched to 1.6.7. Then a week or so later I retried 1.6.10 and everything worked fine. I then used 1.6.10 without problems during at least two weeks, switched to 1.6.11 without problems until suddenly the error appeared after which it was also back in 1.6.10. Rebooting did not help. Deleting and reinstalling 1.6.11 fixed the problem, at least temporarily. When I come across it again I will try to fiddle with memory to see if that helps.

Best regards,

Clemens Valens
Technical Manager

http://www.elektor-labs.com/

E clemens.valens@eimworld.com
T +31 (0)46 4389444      M +33 (0)666404766
F +31 (0)46 4370161     S clemens_valens
P.O. Box 11 – 6114 ZG Susteren – Netherlands

Elektor is part of EIM – Elektor International Media BV
[cid:image002.png@01D2078E.ED2C7570]

[Medienpartner Logo Englisch]
Visit us at electronica:
November 08 – 11, 2016
Booth EOE 200, Entry East

cmaglie, 2016-09-05:
> For me it happened on Windows 7, not XP.

Ah! I missed that you are using W7... well this is new, until now we had reports only for XP.
If you ever get the error again It may be worth investigate if it is exactly the same reported here, or if it is something else and open another issue accordingly with more details.

elektor-labs, 2016-09-05:
In my first post I did mention that I was on W7. Anyway, as far as I can see from the error message, it is exactly the same error, which is why I put it here. Since the toolchain changed in 1.6.10 I would expect the error somewhere in there. I repeat that I have never ever on no other version of the IDE, Windows, Linux or OSX seen this error besides now in 1.6.10 and 1.6.11.

HiTechHiTouch, 2016-09-09:
OK, I got a failure, source, and full dumps all zipped up.  It's BIG -- 125 MB.  What's the best way to pass this to some developer who will review it and pass it to GCC?

Here's a .png screen print of some Windgb output.  I'm not a windows internals person, but I suspect this is a bad pointer from uninitialized data (0xf0f0f0f0f).

[image]

BTW, if the gnucc people could provide symbols....

HiTechHiTouch, 2016-09-09:
I may be overthinking this and getting too subtle, but I wonder if there's some case where XP doesn't provide a clean page, but the GCC compiler expects one.  C requires "uninitialized" data to be zero.  

The program does a malloc (or such) and windows provides a new page, which 99.99% of the time is clean.  But there might not be a contractual obligation in this matter, so very occasionally an old page is reused and isn't zeroed out.

In particular, there are many heap manager variations in windows, and it might actually be the heap manager which is expected (by C malloc et. al.) to clear newly provided memory.  But some combination of heap parameters/heap provider and malloc are not compatible, so C doesn't always get zeroed out memory.

What brings this to mind is that by enabling heap debugging in Gflags, I may have caused heap memory to initialize things to 0xf0f0f0f0 instead of zero, which is the source of the pointer exception I trapped above.

In other words, the dump isn't for our real problem.

HiTechHiTouch, 2016-09-12:
Getting close!

First, running under the debugger always causes memory to be initialized with various patterns.  I suspect this will be a catastrophic collision with GCC's assumption that memory allocated from the system is always zeroed out, thus automatically full-filling the C requirement that "uninitialized" means zero.  

This is going to make debugging in a windows environment almost impossible.  I wonder if the GCC run-time people have never done serious work under windows.

Second, good news! I've gotten an exception where memory contains 0xfeeefeee, which is the value the heap manager places in storage after the heap is freed.

Bad news:  the failing program is now \liblto_plugin-0.dll, a LD.exe plug-in.

I'm afraid this is going to be a matter of rebuilding the AVR tool chain (i.e. Gcc/G++) on my system.

Why can't I find any support from the Arduino IDE people?

Here's the Windgb output:

Microsoft (R) Windows Debugger Version 6.12.0002.633 X86
Copyright (c) Microsoft Corporation. All rights reserved.

CommandLine: "c:/documents and settings/jared/local settings/application data/arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe" "-plugin" "c:/documents and settings/jared/local settings/application data/arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/../libexec/gcc/avr/4.9.2/liblto_plugin-0.dll" "-plugin-opt=c:/documents and settings/jared/local settings/application data/arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/../libexec/gcc/avr/4.9.2/lto-wrapper.exe" "-plugin-opt=-fresolution=C:\TMP\ccV2Ta9A.res" "-plugin-opt=-pass-through=-lgcc" "-plugin-opt=-pass-through=-lm" "-plugin-opt=-pass-through=-lc" "-plugin-opt=-pass-through=-latmega32u4" "-mavr5" "-Tdata" "0x800100" "-o" "C:\TMP\build692751f7cbd22bc67458fdfc887a2b9c.tmp/Ammeter.ino.elf" "c:/documents and settings/jared/local settings/application data/arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/../lib/gcc/avr/4.9.2/../../../../avr/lib/avr5/crtatmega32u4.o" "-LC:\TMP\build692751f7cbd22bc67458fdfc887a2b9c.tmp" "-Lc:/documents and settings/jared/local settings/application data/arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/../lib/gcc/avr/4.9.2/avr5" "-Lc:/documents and settings/jared/local settings/application data/arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/../lib/gcc/avr/4.9.2/../../../../avr/lib/avr5" "-Lc:/documents and settings/jared/local settings/application data/arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/../lib/gcc/avr/4.9.2" "-Lc:/documents and settings/jared/local settings/application data/arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/../lib/gcc" "-Lc:/documents and settings/jared/local settings/application data/arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/../lib/gcc/avr/4.9.2/../../../../avr/lib" "--gc-sections" "C:\TMP\build692751f7cbd22bc67458fdfc887a2b9c.tmp\sketch\Ammeter.ino.cpp.o" "C:\TMP\build692751f7cbd22bc67458fdfc887a2b9c.tmp\libraries\Pins_Jared\I2Cscan.c.o" "C:\TMP\build692751f7cbd22bc67458fdfc887a2b9c.tmp\libraries\NewliquidCrystal\FastIO.cpp.o" "C:\TMP\build692751f7cbd22bc67458fdfc887a2b9c.tmp\libraries\NewliquidCrystal\I2CIO.cpp.o" "C:\TMP\build692751f7cbd22bc67458fdfc887a2b9c.tmp\libraries\NewliquidCrystal\LCD.cpp.o" "C:\TMP\build692751f7cbd22bc67458fdfc887a2b9c.tmp\libraries\NewliquidCrystal\LiquidCrystal.cpp.o" "C:\TMP\build692751f7cbd22bc67458fdfc887a2b9c.tmp\libraries\NewliquidCrystal\LiquidCrystal_I2C.cpp.o" "C:\TMP\build692751f7cbd22bc67458fdfc887a2b9c.tmp\libraries\NewliquidCrystal\LiquidCrystal_I2C_ByVac.cpp.o" "C:\TMP\build692751f7cbd22bc67458fdfc887a2b9c.tmp\libraries\NewliquidCrystal\LiquidCrystal_SI2C.cpp.o" "C:\TMP\build692751f7cbd22bc67458fdfc887a2b9c.tmp\libraries\NewliquidCrystal\LiquidCrystal_SR.cpp.o" "C:\TMP\build692751f7cbd22bc67458fdfc887a2b9c.tmp\libraries\NewliquidCrystal\LiquidCrystal_SR1W.cpp.o" "C:\TMP\build692751f7cbd22bc67458fdfc887a2b9c.tmp\libraries\NewliquidCrystal\LiquidCrystal_SR2W.cpp.o" "C:\TMP\build692751f7cbd22bc67458fdfc887a2b9c.tmp\libraries\NewliquidCrystal\LiquidCrystal_SR3W.cpp.o" "C:\TMP\build692751f7cbd22bc67458fdfc887a2b9c.tmp\libraries\NewliquidCrystal\SI2CIO.cpp.o" "C:\TMP\build692751f7cbd22bc67458fdfc887a2b9c.tmp\libraries\Wire\Wire.cpp.o" "C:\TMP\build692751f7cbd22bc67458fdfc887a2b9c.tmp\libraries\Wire\utility\twi.c.o" "C:\TMP\build692751f7cbd22bc67458fdfc887a2b9c.tmp\libraries\SDL_Arduino_INA3221\SDL_Arduino_INA3221.cpp.o" "C:\TMP\build692751f7cbd22bc67458fdfc887a2b9c.tmp/core\core.a" "-lm" "--start-group" "-lgcc" "-lm" "-lc" "-latmega32u4" "--end-group"

Symbol search path is: srv*c:\symbols*https://msdl.microsoft.com/download/symbols

Executable search path is: 
ModLoad: 00400000 0051b000   image00400000
ModLoad: 7c900000 7c9b4000   ntdll.dll
LDR: Stack trace database size is 100 Mb 
ModLoad: 7c800000 7c8f7000   C:\WINDOWS\System32\kernel32.dll
ModLoad: 77c10000 77c68000   C:\WINDOWS\System32\msvcrt.dll
ModLoad: 7e410000 7e4a1000   C:\WINDOWS\System32\USER32.dll
ModLoad: 77f10000 77f5a000   C:\WINDOWS\System32\GDI32.dll
(418.ea0): Break instruction exception - code 80000003 (first chance)
eax=00141eb4 ebx=7ffdf000 ecx=00000007 edx=00000080 esi=00141f48 edi=00141eb4
eip=7c90120e esp=0111fb20 ebp=0111fc94 iopl=0         nv up ei pl nz na po nc
[... cut ...]
First chance exceptions are reported before any exception handling.
This exception may be expected and handled.
eax=feeefeee ebx=feeefeee ecx=77c2c2e3 edx=00230608 esi=00516ec0 edi=004c4e9a
eip=004526e8 esp=0111fe60 ebp=0111ff78 iopl=0         nv up ei pl nz na pe nc
cs=001b  ss=0023  ds=0023  es=0023  fs=003b  gs=0000             efl=00010206
*** ERROR: Module load completed but symbols could not be loaded for image00400000
image00400000+0x526e8:
004526e8 8b4308          mov     eax,dword ptr [ebx+8] ds:0023:feeefef6=????????

positron96, 2016-09-12:
Hey there! Good work debugging this stuff. 
I guess Arduino IDE devs give no support because this issue is not related to Arduino IDE at all. The best place for this issue to be explored is probably AVR-GCC project. I have no idea where its page is, probably it's a GCC subproject.

facchinm, 2016-09-12:
Hi @HiTechHiTouch ,
thanks for diving deeply into this problem. We've been testing internally on a bunch of XP machines until we were able to reproduce the bug consistently. If you could test the "updated" binary, you can find it here (http://downloads.arduino.cc/tools/avr-gcc-4.9.2-atmel3.5.3-arduino3-i686-mingw32.zip). This build patches `ld` to avoid taking a risky path (read, containing SSE instructions) and adds debug flags, which are stripped by `-s` but seems to help initializing the memory safely. 
If you could test this build it would be really great. Thanks again for the effort!

cmaglie, 2016-09-12:
> I'm afraid this is going to be a matter of rebuilding the AVR tool chain (i.e. Gcc/G++) on my system.

the build scripts used to create the toolchain are here:

https://github.com/arduino/toolchain-avr

launching `package-avr-gcc.bash` from a mingw32 shell should do the work (beware: it could take several hours to complete...).

HiTechHiTouch, 2016-09-12:
@facchinm will work on it.  You seem to be ahead of where I think I will end up going.

@positron96 I'm thinking the Arduino GUI people are just copying this stuff from Atmel.  I look at C:\Program Files\Arduino\hardware\tools\avr\builtin_tools_versions.txt and see
  arduino.avrdude=6.0.1-arduino5
  arduino.avr-gcc=4.8.1-arduino5
and the when I push a "-v" at the collect invocation, I see 4.9.2.  Then I swap over to Atmel Studio and see their current tool chain is also 4.9.2.  Hmmmm.

HiTechHiTouch, 2016-09-12:
@cmaglie Yes, MinGW seems to be the way to go.  I got a precompiled 4.9.2 setup off github but haven't fired it up.  

My Current problem:

mingw sets an environment variable (bash env) from windows environment, and strips off the drive letter from c:/mingw/bin in the PATH variable.

Now I cd out to another drive where I have my sources, run scripts, and (I think) bash looks for mingw/bin on my source drive, not my c: drive.   

Any suggestions? Any site for mingw support, or do people just depend on stackoverflow.com ?

HiTechHiTouch, 2016-09-12:
@facchinm Martino, please email me.  HiTechHiTouch on yahoo mail.

cmaglie, 2016-09-12:
> mingw sets an environment variable (bash env) from windows environment, and strips off the drive letter from c:/mingw/bin in the PATH variable.

uhm, I never had this problem since I have only one drive C:.

I'm not really a mingw expert, but maybe it's worth trying to a have a look at the `"C:\msys\1.0\etc\fstab"` file? on my Windows box `C:/mingw` is mapped to `/mingw`, maybe changing this will help in your case. (see http://www.mingw.org/wiki/msys for details)

HiTechHiTouch, 2016-09-13:
@cmaglie After I installed cygwin, the problem "went away".  I suspect stuff that should be coming out of mingw/bin may actually be coming out of cygwin/bin.  cygwin is explicit about being first in PATH, and mingw said to append to the end of PATH.

I don't like running with mixed modules, i..e. GCC 4.9.1 in mingw and 5.2.0 in cygwin.  But so far, no really obvious problems.

PS: Sent you email at arduino.cc.  If you didn't get it, please email me (see above).

HiTechHiTouch, 2016-09-22:
Still working, but the build process has become a real swamp of alligators.

ZFix2000, 2016-10-13:
I had the same problem. I found that by including the WIRE library it compiled fine.
#include<wire.h>

sppnk, 2016-11-03:
Compiling a simple code I have found 1.6.12 Arduino IDE version  still have the issue.

Adding or removing some functions it apperas/dissapears.

sppnk, 2016-11-03:
...and change ld.exe with the ld.exe from 1.0.5 or 1.0.6 doesnt work, as old ld.exe's cannot read 'plugin' extra commands and exits with another error.

jk987, 2016-12-13:
I have:
WinXP SP3 (modified in regedit to POSReady to get updates)
Arduino IDE 1.6.13
Digispark board with ATTiny85
(I also have some chineese copy of Arduino Nano)

And today I even couldn't compile empty sketch like this:
```
void setup() {
}
void loop() {
}
```
The error is the same:
`collect2.exe: error: ld returned 5 exit status`

Last line in log before error was:
`"c:\Program Files\Arduino\portable\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-gcc" -Os -Wl,--gc-sections -mmcu=attiny85 -o "c:\tmp/tiny_lowpower.ino.elf" "c:\tmp\sketch\tiny_lowpower.ino.cpp.o" "c:\tmp\core\pins_arduino.c.o" "c:\tmp\core\core.a" "-Lc:\tmp" -lm
`
From this you can see that I already tried the trick with changing "build.path" in "preferences.txt" and also creating "portables" subdirectory. Both these tricks should avoid using stupid AppData directory which in my native language contains diacritics and spaces and lot of other software doesn't handle it correctly.

What helped me was to move Arduino IDE out of "c:\program files\arduino" to "c:\arduino". That's quite common trick to make older software work in Windows7 but I didn't expect that it will make new Arduino SW work in Windows XP.

Sporotrix, 2016-12-17:
Hi, just my 5 cents of contribution. I had the same problem as everyone, Windows XP, IDE 1.6.13 and crazy compilation erros. As the error sugested something related to my board, I decided to downgrade the board specs (Tools > Board > board manager > find your board > select older version > install). I went from "Arduino AVR boards" 1.6.15 back to 1.6.2 and things started working. Didn't noticed any functionality losses or any other errors/bugs with the downgrade. Will try to go up a little to see if thigs stay working.

sdikiy, 2016-12-18:
+1
Windows XP, IDE 1.6.13
`collect2.exe: error: ld returned 5 exit status`
`exit status 1`
`Error compiling for board Arduino Nano.`

...same sketch on Win7(64), IDE 1.6.13 - All is OK!

cmaglie, 2017-01-18:
Can someone check if the build provided here
https://github.com/arduino/Arduino/pull/5863#issuecomment-273129529
may solve the issue?

Earl-deLaFere, 2017-03-10:
Hi All!

I confirm that "Update bundled tools # 5863" solves this problem for me.
Earlier, even the compilation of a very simple sketch, such as Blink, caused the error "collect2.exe: error: ld returned 5 exit status".
After installing the nightly build at 08.03.2017, the compilation of this simple sketch and 14 different ones was successful.

My test environment: Windows XP SP3, CPU AMD Athlon (tm) XP 2400+, Genuino UNO.

Because My CPU is AMD Athlon XP, I would be glad if in the next release the utility arduino-builder, and other utilities written on GO, would be compiled with the parameter GO386 = 387 disabling SSE2. Otherwise, I have to recompile arduino-builder from the source code myself.

cmaglie, 2017-03-30:
Fixed on IDE 1.8.2 (and AVR core 1.6.18).

closed 2017-03-30 by cmaglie