arduino/Arduino #2993: Can't connect to UNO from OS X

Skammi, 2015-04-18:
To Start
I have several mac's late last year the boot loader on my Arduino-Robot (the one from Arduino.CC) got corrupted due to connection issues this on my MacBook Pro. I recently fixed that using the ArduinoISP programmer running under Windows 7 on a BootCamp portion on my iMac.
So i'm a bit suspicious connecting my Arduino-Robot to the MAC OS X and thought with 1.63 I would run some tests using my MacBook Pro (Retina, 15-inch, Late 2013) running OS X 10.10.3 and an Arduino-UNO and the Blink Sketch

1 Running Under OS X. I get the following error message:

COPY >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Arduino: 1.6.3 (Mac OS X), Board: "Arduino Uno"

Sketch uses 1,030 bytes (3%) of program storage space. Maximum is 32,256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2,039 bytes for local variables. Maximum is 2,048 bytes.
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf -v -patmega328p -carduino -P/dev/cu.usbmodem1451 -b115200 -D -Uflash:w:/var/folders/st/1zwpqrs51y78wnn5ygksysjm0000gp/T/build1874575217809810440.tmp/Blink.cpp.hex:i 

avrdude: Version 6.0.1, compiled on Mar 30 2015 at 12:15:29
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/Users/jacob/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem1451
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: ser_open(): can't open device "/dev/cu.usbmodem1451": Resource busy
ioctl("TIOCMGET"): Inappropriate ioctl for device

avrdude done.  Thank you.

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< PAST

2 Running under Windows 7 in bootcamp.
No issues, same board, same USB cable, same USB port, Same Sketch (shared using DropBox).

3.Different Uno board under OS X

COPY >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Arduino: 1.6.3 (Mac OS X), Board: "Arduino Uno"

Sketch uses 1,030 bytes (3%) of program storage space. Maximum is 32,256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2,039 bytes for local variables. Maximum is 2,048 bytes.
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf -v -patmega328p -carduino -P/dev/cu.usbserial-AD021A7D -b115200 -D -Uflash:w:/var/folders/st/1zwpqrs51y78wnn5ygksysjm0000gp/T/build7496004744732684893.tmp/Blink.cpp.hex:i 

avrdude: Version 6.0.1, compiled on Mar 30 2015 at 12:15:29
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/Users/jacob/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbserial-AD021A7D
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: ser_open(): can't open device "/dev/cu.usbserial-AD021A7D": Resource busy
ioctl("TIOCMGET"): Inappropriate ioctl for device

avrdude done.  Thank you.

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< PAST

4.Another Uno board (DFRduino RoMeo) under OS X

COPY >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Arduino: 1.6.3 (Mac OS X), Board: "Arduino Uno"

Sketch uses 1,030 bytes (3%) of program storage space. Maximum is 32,256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2,039 bytes for local variables. Maximum is 2,048 bytes.
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf -v -patmega328p -carduino -P/dev/cu.usbmodem1451 -b115200 -D -Uflash:w:/var/folders/st/1zwpqrs51y78wnn5ygksysjm0000gp/T/build7496004744732684893.tmp/Blink.cpp.hex:i 

avrdude: Version 6.0.1, compiled on Mar 30 2015 at 12:15:29
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/Users/jacob/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem1451
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: ser_open(): can't open device "/dev/cu.usbmodem1451": Resource busy
ioctl("TIOCMGET"): Inappropriate ioctl for device

avrdude done.  Thank you.

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< PAST

Bye the way I have use all of these boards on my 2009 iMac running OS X 10.9.* and Arduino 1.05 without any problems. I replaced the iMac earlier this year and have not been able to Upload to Arduino boards under OS X!

So it seems to me that the connection to the USB port under OS X on Newer Mac's is not robust!

Please advise for a solution as I can't get my Windows7 to my iMac as Windows 7 doesn't recognize the USB 3 Controllers in the iMac

Thanks
Jacob

jrickard, 2015-04-19:
I have the same problem with a January 2014 MacPro and OSX 10.10.2

Struggling to find an explanation for this.

Skammi, 2015-04-20:
Did some further testing.
Borrowed my wife her MacBook Air (11-inch, Late 2010). With OS X 10.10.3 and Arduino 1.6.3

And it WORKED!
So that would bring me to the conclusion that something in the Arduino software is not compatible with the USB hardware in the newer mac's.

Thanks Jacob

Here the results:

COPY >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Build options changed, rebuilding all

Sketch uses 1,030 bytes (3%) of program storage space. Maximum is 32,256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2,039 bytes for local variables. Maximum is 2,048 bytes.
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf -v -patmega328p -carduino -P/dev/cu.usbmodem411 -b115200 -D -Uflash:w:/var/folders/g7/5mf0b3c56j3cp91nsqdzqq3w0000gp/T/build7008513921472331357.tmp/Blink.cpp.hex:i 

avrdude: Version 6.0.1, compiled on Mar 30 2015 at 12:15:29
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/Users/jacob/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem411
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 3.3
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f
avrdude: reading input file "/var/folders/g7/5mf0b3c56j3cp91nsqdzqq3w0000gp/T/build7008513921472331357.tmp/Blink.cpp.hex"
avrdude: writing flash (1030 bytes):

Writing | ################################################## | 100% 0.21s

avrdude: 1030 bytes of flash written
avrdude: verifying flash memory against /var/folders/g7/5mf0b3c56j3cp91nsqdzqq3w0000gp/T/build7008513921472331357.tmp/Blink.cpp.hex:
avrdude: load data flash data from input file /var/folders/g7/5mf0b3c56j3cp91nsqdzqq3w0000gp/T/build7008513921472331357.tmp/Blink.cpp.hex:
avrdude: input file /var/folders/g7/5mf0b3c56j3cp91nsqdzqq3w0000gp/T/build7008513921472331357.tmp/Blink.cpp.hex contains 1030 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.21s

avrdude: verifying ...
avrdude: 1030 bytes of flash verified

avrdude done.  Thank you.
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< PAST

Strikingwolf, 2015-05-09:
Does anyone have an idea when this will be fixed? Preventing me from doing anything with arduino :/

ffissore, 2015-09-25:
Please consult the [arduino forum](http://forum.arduino.cc/) where lots of users reports similar issues

closed 2015-09-25 by ffissore

JacobB05, 2020-08-12:
I have the exact same problem, except that uploading code USED to work, but doesn't anymore.