arduino/Arduino #2997: Compiling/linking error for 3rd party boards

NicoHood, 2015-04-18:
I'll explain the issue with an example (ide 1.6.3, win8):

Install the HID project into /sketchbook/hardware/HID
https://github.com/NicoHood/HID

Restart the IDE
Now select the new board that pops up under tools/boards/leonardo HID Project

Type Consumer.begin() in the empty sketchs setup()
compile the code (Control + R) -> Works fine

Now open the example via File>Sketchbook>hardware>HID>HID>Consumer
It wont compile.

I've tested:
* Add an #error to the Consumer.cpp file -> recognized correct
* Include the file via #include "USB-Core/Consumer.cpp" -> compiling correct.

It seems that the files are not linked correct, maybe due to a subfolder?

The error occurs when:
* Open the example via double click
* Open it via file

It does not occur if you:
* Paste the Code into the window that is opened by the startup
* Or the windows that is generated by a "new sketch" button press

It does not matter which Board was selected at closing.

matthijskooijman, 2015-04-20:
Weird bug. You should probably include the full compiler output here as well?

NicoHood, 2015-04-20:
Not Working:
```
C:\Users\Nico\AppData\Roaming\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino2/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_MANUFACTURER= -DUSB_PRODUCT="Arduino Leonardo" -IC:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid -IC:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\variants\leonardo_hid C:\Users\Nico\AppData\Local\Temp\build8793008032484503174.tmp\Consumer.cpp -o C:\Users\Nico\AppData\Local\Temp\build8793008032484503174.tmp\Consumer.cpp.o 
C:\Users\Nico\AppData\Roaming\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino2/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_MANUFACTURER= -DUSB_PRODUCT="Arduino Leonardo" -IC:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid -IC:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\variants\leonardo_hid C:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid\hooks.c -o C:\Users\Nico\AppData\Local\Temp\build8793008032484503174.tmp\hooks.c.o 
C:\Users\Nico\AppData\Roaming\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino2/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_MANUFACTURER= -DUSB_PRODUCT="Arduino Leonardo" -IC:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid -IC:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\variants\leonardo_hid C:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid\WInterrupts.c -o C:\Users\Nico\AppData\Local\Temp\build8793008032484503174.tmp\WInterrupts.c.o 
C:\Users\Nico\AppData\Roaming\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino2/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_MANUFACTURER= -DUSB_PRODUCT="Arduino Leonardo" -IC:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid -IC:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\variants\leonardo_hid C:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid\wiring.c -o C:\Users\Nico\AppData\Local\Temp\build8793008032484503174.tmp\wiring.c.o 
C:\Users\Nico\AppData\Roaming\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino2/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_MANUFACTURER= -DUSB_PRODUCT="Arduino Leonardo" -IC:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid -IC:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\variants\leonardo_hid C:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid\wiring_analog.c -o C:\Users\Nico\AppData\Local\Temp\build8793008032484503174.tmp\wiring_analog.c.o 
C:\Users\Nico\AppData\Roaming\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino2/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_MANUFACTURER= -DUSB_PRODUCT="Arduino Leonardo" -IC:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid -IC:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\variants\leonardo_hid C:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid\wiring_digital.c -o C:\Users\Nico\AppData\Local\Temp\build8793008032484503174.tmp\wiring_digital.c.o 
C:\Users\Nico\AppData\Roaming\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino2/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_MANUFACTURER= -DUSB_PRODUCT="Arduino Leonardo" -IC:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid -IC:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\variants\leonardo_hid C:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid\wiring_pulse.c -o C:\Users\Nico\AppData\Local\Temp\build8793008032484503174.tmp\wiring_pulse.c.o 
C:\Users\Nico\AppData\Roaming\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino2/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_MANUFACTURER= -DUSB_PRODUCT="Arduino Leonardo" -IC:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid -IC:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\variants\leonardo_hid C:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid\wiring_shift.c -o C:\Users\Nico\AppData\Local\Temp\build8793008032484503174.tmp\wiring_shift.c.o 
C:\Users\Nico\AppData\Roaming\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino2/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_LEONARDO -DARDUINO [... cut ...] eep 
C:\Users\Nico\AppData\Roaming\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino2/bin/avr-objcopy -O ihex -R .eeprom C:\Users\Nico\AppData\Local\Temp\build8793008032484503174.tmp/sketch_apr20a.cpp.elf C:\Users\Nico\AppData\Local\Temp\build8793008032484503174.tmp/sketch_apr20a.cpp.hex 

Sketch uses 4,634 bytes (16%) of program storage space. Maximum is 28,672 bytes.
Global variables use 135 bytes (5%) of dynamic memory, leaving 2,425 bytes for local variables. Maximum is 2,560 bytes.
```

matthijskooijman, 2015-04-20:
The problem is in the Consumer.ino filename. This results in a compiled Consumer.cpp.o file, directly inside the tmp directory. However, the Consumer.cpp file from the core is also compiled to the same Consumer.cpp.o file.

Having the .o filenames conflict in this way should be fixed in the IDE. I think the core files should be compiled inside a subfolder of the tmp folder ("core", or perhaps named after the core itself). This would reduce the potential for conflicts, only when you have a library called "core" this could still conflict. To fix that, a library should be compiled inside a tmp/libraries/name_of_library folder perhaps.

Also interesting is that any subdirectories in the core directory are lost. cpp files in subdirectories are compiled, but the .o path doesn't contain the subdirectory anymore. So if the same filename occurs in the core twice (in different (sub)folders), this would again conflict.

NicoHood, 2015-04-20:
Thx! And I've noted it somewhere in my Documentation to not do this. X_x

Thanks!

Ergo: Someone should add a patch to check the filename vs any compiled cpp?

matthijskooijman, 2015-04-20:
Also note that this is not specific to 3rd party cores - If I create a Print.ino sketch, I expect the IDE will break using the normal Arduino core as well.

PaulStoffregen, 2015-04-20:
I was wondering if this problem would ever come up.... been watching for it for years!

ffissore, 2015-04-22:
ACK. Solving this will also help solving #1337 as we'll be able to copy sketch source files into temp dir.

cmaglie, 2015-04-22:
There is already the sub-folder compile for libraries:

https://github.com/arduino/Arduino/blob/master/arduino-core/src/processing/app/debug/Compiler.java#L997

```Java
  private void recursiveCompileFilesInFolder(File srcBuildFolder, File srcFolder, List<File> includeFolders) throws RunnerException, PreferencesMapException {
    compileFilesInFolder(srcBuildFolder, srcFolder, includeFolders);
    for (File subFolder : srcFolder.listFiles(new OnlyDirs())) {
      File subBuildFolder = new File(srcBuildFolder, subFolder.getName());
      recursiveCompileFilesInFolder(subBuildFolder, subFolder, includeFolders);
    }
  }
```

it uses the `compileFilesInFolder` method to recreate the library's folders structure in the build folder. We can use the same method for compiling the core in a subfolder, say, `arduino-core`. We must also carefully check how much the configurations in `platform.txt` relies on the position of those files.

matthijskooijman, 2015-04-22:
Sounds good (though we'd have to modify the .a file generation as well I believe).

Furthermore, it might be useful to modify [this line](https://github.com/arduino/Arduino/blob/master/arduino-core/src/processing/app/debug/Compiler.java#L973) to insert a fixed "libraries" folder (e.g. so a Foo library is compiled in build_dir/libraries/Foo). This completely rules out any conflict by weirdly named libraries.

Not sure about platform.txt reliability on file position - I think at least the recipes only deal with a single file, regardless of where it is?

NicoHood, 2015-04-22:
If you are changing things in the library linkage, would it also be possible to implement this option with the fix as well? https://github.com/arduino/Arduino/issues/2800

PaulStoffregen, 2015-04-22:
That's a very different issue, unrelated to simply building the core library in its own directory for the purpose of avoiding file naming conflicts.

commit 5a35be06 (2015-06-29): Compiler: cores files get compiled under "core", libraries under "libraries". Fixes #2997
closed 2015-06-29 by ffissore