arduino/Arduino #2998: Why -g is included in the compilation sequence?

anonymouse64, 2015-04-18:
Hello, 
I am wondering why the Arduino software always compiles with the -g (debugging in native output) option. 
Background:
I am using the version of avr-gcc (and other tools) included in the Arduino software release, and before 1.6.3, I didn't have any problem compiling without -g, but now I get a linker returning 255 or 5 (or other random numbers sometimes) error. Including -g with Arduino 1.6.3 causes it to compile fine. When I switch to using 1.6.1 or 1.6.0 (I haven't tried 1.6.2), the issue doesn't appear. Strangely it seems as though including the -g option causes the linker to find files it should have found anyways. I can post more details (including verbose compilation output) if necessary. (I am using Windows 8.1 for reference)
I realize my specific issue is likely with avr-gcc and related tools itself, but it raised my curiosity as to why -g is always included when compiling on Arduino.
Thanks,
Ian

matthijskooijman, 2015-04-20:
I'm a bit confused. You say adding -g fixes a compilation issue, but also that -g is always included already. Those seem contradictory to me?

Posting verbose compilation output, preferable for both the working and failing cases would be useful. Posting the failing sketch code (preferable with as much code as possible removed) would also be useful.

anonymouse64, 2015-04-20:
Sorry, perhaps I should have been clearer. I am using the avr-gcc tools and utilities (like avrude, avr-gcc, etc.) to upload to an Arduino board without using the Java-based Arduino software. When I do this, I have never included -g, and it worked fine without -g, until I updated to version 1.6.3. Now, the compilation will "fail" when -g is omitted. Interestingly enough, it still produces the elf output.

I noticed that the Arduino Java software includes the -g option always when compiling and I was wondering why this is, and if there is a reason that debugging is turned on for the compilation sequence.

To reproduce the issue inside the Arduino Java software, open up platforms.txt inside (avr install location)/hardware/arduino/avr, and remove -g from all the compilation recipes for "compiler.c.flags", "compiler.cpp.flags" and from "compiler.s.flags".

Now, open up the Arduino Java software and try to compile just about any sketch (including a blank one as I have done so below), and it will fail due to a linker issue. The sketch I tried compiling below is as follows:
```
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

This is the output from the Arduino software

```
Arduino: 1.6.3 (Windows 8.1), Board: "Arduino Uno"

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++ -c -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard C:\Users\Ian\AppData\Local\Temp\build4408645479113764614.tmp\sketch_apr20a.cpp -o C:\Users\Ian\AppData\Local\Temp\build4408645479113764614.tmp\sketch_apr20a.cpp.o 

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc -c -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\hooks.c -o C:\Users\Ian\AppData\Local\Temp\build4408645479113764614.tmp\hooks.c.o 

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc -c -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\WInterrupts.c -o C:\Users\Ian\AppData\Local\Temp\build4408645479113764614.tmp\WInterrupts.c.o 

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc -c -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\wiring.c -o C:\Users\Ian\AppData\Local\Temp\build4408645479113764614.tmp\wiring.c.o 

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc -c -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\wiring_analog.c -o C:\Users\Ian\AppData\Local\Temp\build4408645479113764614.tmp\wiring_analog.c.o 

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc -c -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\wiring_digital.c -o C:\Users\Ian\AppData\Local\Temp\build4408645479113764614.tmp\wiring_digital.c.o 

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc -c -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\wiring_pulse.c -o C:\Users\Ian\AppData\Local\Temp\build4408645479113764614.tmp\wiring_pulse.c.o 

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc -c -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\wiring_shift.c -o C:\Users\Ian\AppData\Local\Temp\build4408645479113764614.tmp\wiring_shift.c.o 

[... cut ...]
(c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr5\libc.a)strrev.o
(c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr5\libc.a)errno.o
(c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/avr5\libgcc.a)_mulshisi3.o
(c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/avr5\libgcc.a)_prologue.o
(c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/avr5\libgcc.a)_epilogue.o
```

anonymouse64, 2017-02-09:
I think that this problem is related to/explained by https://github.com/arduino/Arduino/issues/5539

Is that correct? Is the -g option used so all necessary information is included in the elf?

matthijskooijman, 2017-02-10:
Looking at the compiler output, this could very well be a compiler bug (given that the linker exits without any error message). @anonymouse64, is this problem still occuring for you with the most recent version? I think we might have upgraded the compiler version since then.

As for the issue you link to, that might indeed solve this issue, if the compiler is somehow confused by having debugging symbols in the object files without -g being passed to the linker.

As for why -g is passed in the first place, that allows using an (external) debugger, or disassembling the binary and dumping the related source code in between (though I've seen that LTO messes up the correctness of the debugging symbols in some places).

PaulStoffregen, 2017-02-10:
The -g debub symbols also allows the compiler to show more meaningful error messages.

anonymouse64, 2017-02-11:
Yes, I have confirmed that it now works with the updated toolchain in Arduino 1.8.1.
Thanks for the explanation!

closed 2017-02-11 by anonymouse64