arduino/Arduino #3001: arduino software should use original driver instead of apple one

LunarLanding, 2015-04-19:
If this software is expected to work out of the box, the procedure described here:
http://forum.arduino.cc/index.php?topic=198539.30
should be applied automatically for osx 10.9+

ffissore, 2015-04-23:
I can't find the procedure you mentioned. Can you be more specific? Also, we don't bundle drivers for macosx. And our getting started guide links to ftdi website http://www.arduino.cc/en/Guide/MacOSX#toc3

LunarLanding, 2015-04-25:
My bad, I linked to somepage of the thread instead of the OP:
http://forum.arduino.cc/index.php?topic=198539.0
But since the guide links to the ftdi drivers, I guess it's a moot point
... but OSX does have a driver that partially works with Arduino, which is a source of confusion, guide or no guide.

closed 2015-04-29 by ffissore