arduino/Arduino #3003: Non portable shebang for bash

pgiffuni, 2015-04-19:
#!/bin/bash is very bad idea.
In essence some platforms (like BSD or Solaris don't have bash in that place so this will cause issues. You instead should use 
#!/bin/env bash

Also Ubuntu has started a trend of avoiding bash since the standard shell (dash) is faster so it's better to use the more portable /bin/sh when possible. See:
https://wiki.ubuntu.com/DashAsBinSh

closed 2015-05-29 by ffissore

pgiffuni, 2015-05-29:
Thanks ... 

I know it's not urgent but it looks like you only changed one line, while my patch reveals at least 4 occurrences of /bin/bash:
arduino-core/src/processing/app/i18n/pull.sh
arduino-core/src/processing/app/i18n/push.sh
build/linux/dist/arduino
build/build_pull_request.bash

It doesn't hurt to fix them all.
(The first 2 work with plain /bin/sh)

ffissore, 2015-05-29:
Sorry, I haven't looked at your patched, I've missed it. Please, provide a pull request with all the necessary files changed

pgiffuni, 2015-05-29:
I am not a heavy user of git/github (yet). 
Please just change /bin/bash for /usr/bin/env bash on the mentioned files.
That should be enough to get this building on FreeBSD.

commit 6974978e (2015-05-29): Linux: arduino startup script uses #!/usr/bin/env bash. Fixes #3003