arduino/Arduino #3006: Provide sources for drivers

bromagosa, 2015-04-20:
I couldn't find the source code for these drivers:

https://github.com/arduino/Arduino/tree/06d95d42257c606a1c8c2e8f76074c60db9974c6/build/windows/dist/drivers

Are those proprietary?

Thanks! :)

facchinm, 2015-04-20:
Hi @bromagosa ,

that driver leverage on the standard Windows USB-CDC driver. 
The `.inf` file _is_ the driver itself as it tells the OS that certain VID/PID combinations can use the CDC standard driver. 

If you have a more specific question (like the signing procedure) feel free to reopen the issue :wink: 

Cheers

closed 2015-04-20 by facchinm

bromagosa, 2015-04-20:
Oh, I see! That goes to show I know nothing about Windows...

Are those drivers under the GPL too? Can we redistribute them?

PaulStoffregen, 2015-04-20:
The actual driver code is in USBSER.SYS, which comes with Windows.  If you want the source code for that, you'll have to contact Microsoft!

bromagosa, 2015-04-21:
I understand that, yes. But can we redistribute the files under https://github.com/arduino/Arduino/tree/06d95d42257c606a1c8c2e8f76074c60db9974c6/build/windows/dist/drivers ? What is the license for those?

cmaglie, 2015-06-17:
@bromagosa 
the license on the `arduino.inf` drivers has been set.

bromagosa, 2015-06-17:
Cool, thanks! :)

commit e73cae6f (2015-06-10): Fixed license for drivers

Close #3006