arduino/Arduino #3007: Change error copy in Library/Board Managers

00alis, 2015-04-20:
Copy should be "Unable to reach Arduino.cc due to possible network issues"

instead of
"No internet connection available...." as in this screenshot
[image]

@ffissore

ffissore, 2015-04-22:
Should the whole phrase read "Unable to reach Arduino.cc due to possible network issues. The list of available boards is not complete. You will be able to manage only the boards you've already installed." or just "Unable to reach Arduino.cc due to possible network issues" ?

00alis, 2015-04-22:
@ffissore please refer to this conversation on Basecamp, this is what has been decided. Thanks!
https://basecamp.com/2546761/projects/5094615/messages/40584564

commit 269a8b11 (2015-04-22): "No internet connection" copy change. Closes #3007
closed 2015-04-22 by ffissore