arduino/Arduino #3009: High CPU usage

gvuyk, 2015-04-20:
Windows 7 64
Arduino IDE 1.6.3
Process: "listComPorts.exe" 
It's even more noticeable when having multiple instances running as each one spawns its own listComPorts.exe

feveran, 2015-04-27:
I got same issue with WMI Provider Host when running arduino ide 1.6.0 and newer versions in high performance or balanced mode (on aptop) but not in power saver mode.

asantaga, 2015-06-14:
ditto.. makes compiling incredibly slow

ffissore, 2015-09-25:
Last week, on our windows box, the IDE began to take lots of time to start: it stay stuck to "loading configuration..." for much longer than usual. Guess what? It's the antivirus, Avira. Once turned it off, everything started being waayyy faster

closed 2015-09-25 by ffissore

phme, 2015-10-01:
I'm still having the issue, antivirus disabled. WMI Provider Host (WmiPrvSE.exe) at stable 25% CPU as soon as the IDE is launched. Something about the way USB port is being polled?
Windows 8.1
Arduino IDE 1.6.6 Hourly 2015/10/01 10:54

tangshuran, 2015-10-19:
i got the same problem

tangshuran, 2015-10-19:
any solution?

phme, 2015-10-19:
Well, this issue is still marked as closed, so I suspect the problem is not that interesting.
On my machine, it's still manageable, as it maxes out one core out of four. So I just have to put up with annoying fan noise, but otherwise it works.
I guess the long-term solution would be to use Visual Studio or Atmel Studio, if you're on PC.

tangshuran, 2015-10-19:
thanks a lot. It just makes me confused that what kind of service cost 20% ability of my i7 core. And the machine really get hot.
My crude solution is shutting down the arduino IDE immediately after  copying examples and compiling.

ckpinguin, 2015-11-06:
Well after all, why has Arduino to use WMI Host fully when idle? It heatens up my notebook making it sound like a starting airplane

ffissore, 2015-11-06:
indeed it doesn't, not intentionally. it's also not a widespread issue: looks like only some windows are affected

HopWorks, 2016-01-29:
I just chased this to Arduino 1.6.5 also. Having the same issue with WMI Provider Host on a Windows 10 machine.

facchinm, 2016-02-02:
Hi @HopWorks ,
can you test if WMI Provider Host cpu occupation stays high with the hourly build? 
It should have been solved by #4211

HopWorks, 2016-02-02:
I will test with the new build, but FYI... the link on the page you sent me is broken.
[http://downloads.arduino.cc/javaide/pull_requests/arduino-PR-4211-BUILD-475-windows.zip](url)

facchinm, 2016-02-02:
Yep, the builds get canceled when they are merged so it is expected.
If you could test it using the nightly build it would be great!