arduino/Arduino #3010: Couldn't determine program size, Linux, x64

neo2001, 2015-04-20:
Since upgrading from 1.6.2 to 1.6.3, every time I want to verify or upload a sketch I get the following error(s):

Couldn't determine program size: /home/xxxxx/arduino/hardware/tools/avr/bin/avr-size: '/tmp/build5063569465113905799.tmp/Blink.cpp.elf': No such file

avrdude: can't open input file /tmp/build5063569465113905799.tmp/Blink.cpp.hex: No such file or directory
avrdude: read from file '/tmp/build5063569465113905799.tmp/Blink.cpp.hex' failed

I tried two different systems, both running with Ubuntu 14.04 LTS.

Only solution for me was to downgrade back to 1.6.2.

matthijskooijman, 2015-04-21:
Could you post the full (verbose) output? The errors you are showing seem to be the result of compilation failing somehow, so I expect there are more meaninful errors further up in the output.

neo2001, 2015-04-21:
No problem at all:
```
/home/xxxxx/bin/arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino -I/home/xxxxx/bin/arduino/hardware/arduino/avr/variants/eightanaloginputs /tmp/build1478632555417280387.tmp/Blink.cpp -o /tmp/build1478632555417280387.tmp/Blink.cpp.o 
/home/xxxxx/bin/arduino/hardware/tools/avr/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino -I/home/xxxxx/bin/arduino/hardware/arduino/avr/variants/eightanaloginputs /home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino/hooks.c -o /tmp/build1478632555417280387.tmp/hooks.c.o 
/home/xxxxx/bin/arduino/hardware/tools/avr/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino -I/home/xxxxx/bin/arduino/hardware/arduino/avr/variants/eightanaloginputs /home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino/wiring_analog.c -o /tmp/build1478632555417280387.tmp/wiring_analog.c.o 
/home/xxxxx/bin/arduino/hardware/tools/avr/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino -I/home/xxxxx/bin/arduino/hardware/arduino/avr/variants/eightanaloginputs /home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino/wiring.c -o /tmp/build1478632555417280387.tmp/wiring.c.o 
/home/xxxxx/bin/arduino/hardware/tools/avr/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino -I/home/xxxxx/bin/arduino/hardware/arduino/avr/variants/eightanaloginputs /home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino/wiring_digital.c -o /tmp/build1478632555417280387.tmp/wiring_digital.c.o 
/home/xxxxx/bin/arduino/hardware/tools/avr/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino -I/home/xxxxx/bin/arduino/hardware/arduino/avr/variants/eightanaloginputs /home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino/WInterrupts.c -o /tmp/build1478632555417280387.tmp/WInterrupts.c.o 
/home/xxxxx/bin/arduino/hardware/tools/avr/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino -I/home/xxxxx/bin/arduino/hardware/arduino/avr/variants/eightanaloginputs /home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino/wiring_pulse.c -o /tmp/build1478632555417280387.tmp/wiring_pulse.c.o 
/home/xxxxx/bin/arduino/hardware/tools/avr/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino -I/home/xxxxx/bin/arduino/hardware/arduino/avr/variants/eightanaloginputs /home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino/wiring_shift.c -o /tmp/build1478632555417280387.tmp/wiring_shift.c.o 
/home/xxxxx/bin/arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino -I/home/xxxxx/bin/arduino/hardware/arduino/avr/variants/eightanaloginputs /home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino/Print.cpp -o /tmp/build1478632555417280387.tmp/Print.cpp.o 
/home/xxxxx/bin/arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino -I/home/xxxxx/bin/arduino/hardware/arduino/avr/variants/eightanaloginputs /home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino/HardwareSerial3.cpp -o /tmp/build1478632555417280387.tmp/HardwareSerial3.cpp.o 
/home/xxxxx/bin/arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino -I/home/xxxxx/bin/arduino/hardware/arduino/avr/variants/eightanaloginputs /home/xxxxx/bin/arduino/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp -o /tmp/build1478632555417280387.tmp/HardwareSerial.cpp.o 
/home/xxxxx/bin/arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/xxxxx/bin/arduino/hardware/arduino/avr [... cut ...]
/home/xxxxx/bin/arduino/hardware/tools/avr/bin/avr-objcopy -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 /tmp/build1478632555417280387.tmp/Blink.cpp.elf /tmp/build1478632555417280387.tmp/Blink.cpp.eep 
/home/xxxxx/bin/arduino/hardware/tools/avr/bin/avr-objcopy -O ihex -R .eeprom /tmp/build1478632555417280387.tmp/Blink.cpp.elf /tmp/build1478632555417280387.tmp/Blink.cpp.hex 
Couldn't determine program size: /home/xxxxx/bin/arduino/hardware/tools/avr/bin/avr-size: '/tmp/build1478632555417280387.tmp/Blink.cpp.elf': No such file
```

mschlenker, 2015-04-21:
Did you run out of space in your $TEMP?

neo2001, 2015-04-25:
This error/problem seems to be very odd.

On one system it disappeared from one day to another after I tried the Arduino CLI - even though I didn't change anything and tried at least a dozen times before (incl. deleting userdir, reboot system etc..). But now CLI and GUI are working as expected (using the last official 1.6.3 release).

On the other system the problem was more stubborn, but is now gone, too. After tinkering around a little bit, I finally downloaded the sources from git and built the Linux x64 package locally. Somehow this solved the problem. Still strange, because the nightlies/"hourlies" 1.6.4 and 1.6.3 *always* got me this error, while 1.6.2 *never* did.

So this issue is - at least for me - somehow solved. :-)

mschlenker, 2015-04-27:
> This error/problem seems to be very odd.

Which filesystem do you use? Are you using SSD or HDD? Could you also reproduce the behaviour when `/tmp` is mounted as `tmpfs`? You might have run into a race condition in the filesystem: The first process that wrote the hex file (the compiler) did not yet completely close it while the second process (avrdude) already tries to read. I remember running in a similar issue in shell scripts with Ubuntu 14.04 LTS a few months ago, I "solved" be sleeping a fifth or tenth of a second between the process writing and that one reading a file.

neo2001, 2015-04-29:
On both systems the root partition (incl. /tmp) is on a SSD using the ext4 filesystem.
But on one of these systems the /home partition is on a HDD - which is also the partition where my Arduino installation is located.

Since my last post I never encountered the issue again. But if I will and find something that helps or changes anything, I will report back.

closed 2015-04-29 by neo2001