arduino/Arduino #3012: arduino 1.6.3 for linux - avrdude missing usb support

lpaseen, 2015-04-21:
Trying to use my usbasp programmer directly from ide but it fails because avrdude missing libusb
Running with distro provided avrdude 6.1 works so libusb is ok, it's just not compiled in to the 1.6.3 embedded version of avrdude.

avrdude: Version 6.0.1, compiled on Mar 30 2015 at 11:53:46
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/usr/local/src/peters/Misc/arduino-1.6.3/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/home/user/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

Error while burning bootloader.
         Using Port                    : usb
         Using Programmer              : usbasp
avrdude: error: no usb support. please compile again with libusb installed.
avrdude: programmer operation not supported

avrdude done.  Thank you.

mschlenker, 2015-04-21:
Possibly a duplicate of: https://github.com/arduino/toolchain-avr/issues/23

matthijskooijman, 2015-04-21:
This is a known problem, see https://github.com/arduino/toolchain-avr/issues/23

closed 2015-04-21 by matthijskooijman

ffissore, 2015-04-23:
1.6.4 core and tools (with libusb fix) available through boards manager