arduino/Arduino #3016: Connecting YUN to a Wireless with less than 8char password

NicolaClab, 2015-04-22:
Using the config panel of my Arduino YUN, i cannot connect my company wifi because there is a numeric password length 6char and Arduino tell..password must have 8 or more char.

ffissore, 2015-04-22:
Any chance you can use a 8+ char password? 6 chars is really weak

NicolaClab, 2015-04-22:
Yes, i know, but it's the Company Password, i cannot modify Company policy. How can i workaround?

ffissore, 2015-04-22:
ATM you can switch to the advanced configuration panel and configure wifi from there

NicolaClab, 2015-04-22:
What section to config you to attach it to my wifi?

[image]

ffissore, 2015-04-22:
Please check out this video: https://www.youtube.com/watch?v=yAwnWU5VWD8
Closing as this is turning into a support ticket rather than a bug report.
For further assistance in using the yun, please refer to the dedicated Arduino Forum http://forum.arduino.cc/index.php?board=93.0

closed 2015-04-22 by ffissore

matthijskooijman, 2015-04-22:
FWIW, I think that < 8 character passwords aren't really allowed by the wifi/WPA specification. This minimum isn't very strictly defined it seems, but a lot of software and devices adhere to it. See also http://serverfault.com/a/165102