arduino/Arduino #3022: Problems programming bootloader arduino 2009 to ATtiny 85 and ATtiny 84

glprinamello, 2015-04-22:
Working with arduino board 2009 with new Bootloader and shield for programming ATtiny 44/84 and Attiny25/85

Have problems during programming. Error which appears:

"Arduino:1.6.3 (Windows XP), Scheda:"ATtiny x5, ATtiny85 @ 8 MHz (internal oscillator; BOD disabled)" avr_read(): error reading address 0x0000 read operation not supported for memory "lock" avrdude: failed to read all of lock memory, rc=-2 Errore durante la scrittura del bootloader"

Need Help

mschlenker, 2015-04-23:
Workaround: Try reading the fuses without reading memory lock, the path to avrdude and the config used can be taken from the output of the Arduino IDE:

```
avrdude -C "$AVRCONF" -p t45 -c arduino -b19200 -P /dev/ttyACM0 -U lfuse:r:-:h -U hfuse:r:-:h -U efuse:r:-:h
```

If this is successful you can set the fuses accordingly:

```
avrdude -p t45 -C "$AVRCONF" -c arduino -b19200 -P /dev/ttyACM0 -U lfuse:w:0xE2:m -U hfuse:w:0xDF:m -U efuse:w:0xFF:m
```

Take the fuse settings you need from the `boards.txt` containing your HW definitions. 

I think this problem exists since a while. However I never cared since I use shell scripts to change fuse settings which is quicker than opening the Arduino IDE.

glprinamello, 2015-04-23:
Thank you very much, i shall try this evening as soon as i get home.

The only thing i don't understand if memory is locked how can i program it
in the future?

mschlenker, 2015-04-23:
Related to https://github.com/arduino/Arduino/issues/808

Changing the avrdude.conf for more ATtinys might help.

ffissore, 2015-04-23:
Can this patch help? https://github.com/arduino/toolchain-avr/pull/21 It's available with avr core 1.6.4 (through boards manager)

mschlenker, 2015-04-23:
@ffissore I doubt. The problem seems to berelated to a missing read section in `avrdude.conf, seee #808

```
memory "lock"
         size            = 1;
         write           = "1 0 1 0  1 1 0 0  1 1 1 x  x x x x",
                           "x x x x  x x x x  1 1 i i  i i i i";
        min_write_delay = 9000;
        max_write_delay = 9000;
       ;
```

I'll fix and test this later today at least for the ATtiny85 and 45. I can only test with ATtiny85 and 45, so someone else has to test for other ATtinys.

mschlenker, 2015-04-23:
@glprinamello Would you please provide me with the exact version of boards.txt you are using for the ATtiny?

glprinamello, 2015-04-23:
Nop still have problems but now something is moving.

I changed avrdude.conf file and added missing part for reading but still
gives me error trough IDE.

"Arduino:1.6.3 (Windows XP), Scheda:"ATtiny x5, ATtiny85 @ 8 MHz  (internal
oscillator; BOD disabled)"

C:\Documents and
Settings\Giangy_Home\Desktop\PROGETTAZIONE\ArduinoProg\Arduino\hardware\tools\avr/bin/avrdude
-CC:\Documents and
Settings\Giangy_Home\Desktop\PROGETTAZIONE\ArduinoProg\Arduino\hardware\tools\avr/etc/avrdude.conf
-v -v -v -v -pattiny85 -cstk500v1 -PCOM2 -b19200 -e -Ulock:w:0xFF:m
-Uefuse:w:0xFF:m -Uhfuse:w:0xD7:m -Ulfuse:w:0xE2:m

avrdude: Version 6.0.1, compiled on Mar 30 2015 at 14:56:06

         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/

         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Documents and
Settings\Giangy_Home\Desktop\PROGETTAZIONE\ArduinoProg\Arduino\hardware\tools\avr/etc/avrdude.conf"

avrdude: error at C:\Documents and
Settings\Giangy_Home\Desktop\PROGETTAZIONE\ArduinoProg\Arduino\hardware\tools\avr/etc/avrdude.conf:332:
programmer type must be written as "id_type"

Errore durante la scrittura del bootloader "

glprinamello, 2015-04-23:
Sorry where can i find this information?

2015-04-23 19:22 GMT+02:00 Gian Luca Prina Mello <glprinamello@gmail.com>:

glprinamello, 2015-04-23:
i'm using this file:

2015-04-23 19:26 GMT+02:00 Gian Luca Prina Mello <glprinamello@gmail.com>:

> Sorry where can i find this information?
>
> 2015-04-23 19:22 GMT+02:00 Gian Luca Prina Mello <glprinamello@gmail.com>:
>
>> Nop still have problems but now something is moving.
>>
>> I changed avrdude.conf file and added missing part for reading but still
>> gives me error trough IDE.
>>
>> "Arduino:1.6.3 (Windows XP), Scheda:"ATtiny x5, ATtiny85 @ 8 MHz
>>  (internal oscillator; BOD disabled)"
>>
>> C:\Documents and
>> Settings\Giangy_Home\Desktop\PROGETTAZIONE\ArduinoProg\Arduino\hardware\tools\avr/bin/avrdude
>> -CC:\Documents and
>> Settings\Giangy_Home\Desktop\PROGETTAZIONE\ArduinoProg\Arduino\hardware\tools\avr/etc/avrdude.conf
>> -v -v -v -v -pattiny85 -cstk500v1 -PCOM2 -b19200 -e -Ulock:w:0xFF:m
>> -Uefuse:w:0xFF:m -Uhfuse:w:0xD7:m -Ulfuse:w:0xE2:m
>>
>> avrdude: Version 6.0.1, compiled on Mar 30 2015 at 14:56:06
>>
>>          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
>>
>>          Copyright (c) 2007-2009 Joerg Wunsch
>>
>>          System wide configuration file is "C:\Documents and
>> Settings\Giangy_Home\Desktop\PROGETTAZIONE\ArduinoProg\Arduino\hardware\tools\avr/etc/avrdude.conf"
>>
>> avrdude: error at C:\Documents and
>> Settings\Giangy_Home\Desktop\PROGETTAZIONE\ArduinoProg\Arduino\hardware\tools\avr/etc/avrdude.conf:332:
>> programmer type must be written as "id_type"
>>
>> Errore durante la scrittura del bootloader "
>>
>>
>>
>>> @ffissore <https://github.com/ffissore> I doubt. The problem seems to
>>> berelated to a missing read section in `avrdude.conf, seee #808
>>>
>>> memory "lock"
>>>          size            = 1;
>>>          write           = "1 0 1 0  1 1 0 0  1 1 1 x  x x x x",
>>>                            "x x x x  x x x x  1 1 i i  i i i i";
>>>         min_write_delay = 9000;
>>>         max_write_delay = 9000;
>>>        ;
>>>
>>> I'll fix and test this later today at least for the ATtiny85 and 45. I
>>> can only test with ATtiny85 and 45, so someone else has to test for other
>>> ATtinys.

menu.cpu=Micro

###########################################################################

tinyx4.name=ATtiny x4

tinyx4.bootloader.tool=avrdude
tinyx4.bootloader.unlock_bits=0xFF
tinyx4.bootloader.lock_bits=0xFF
tinyx4.build.board=AVR_TINYX4

tinyx4.menu.cpu.attiny84at16=ATtiny84 @ 16 MHz  (external crystal; 4.3 V BOD)

tinyx4.menu.cpu.attiny84at16.upload.tool=arduino:arduinoisp
tinyx4.menu.cpu.attiny84at16.upload.maximum_size=8192
[... 1074 lines cut ...]
attiny2313at1.upload.tool=arduino:arduinoisp

attiny2313at1.upload.maximum_size=2048
attiny2313at1.upload.maximum_data_size=128

# Default clock (slowly rising power; long delay to clock; 8 MHz internal; divide clock by 8)
# Int. RC Osc. 8 MHz; Start-up time: 14 CK + 65 ms; [CKSEL=0100 SUT=10]; default value
# Divide clock by 8 internally; [CKDIV8=0]
# Brown-out detection disabled; [BODLEVEL=111]
# Serial program downloading (SPI) enabled; [SPIEN=0]
# Preserve EEPROM memory through the Chip Erase cycle; [EESAVE=0]

attiny2313at1.bootloader.low_fuses=0x64
attiny2313at1.bootloader.high_fuses=0x9F
attiny2313at1.bootloader.extended_fuses=0xFF

attiny2313at1.bootloader.path=empty
attiny2313at1.bootloader.file=empty2313at1.hex
attiny2313at1.bootloader.tool=arduino:avrdude

attiny2313at1.build.mcu=attiny2313
attiny2313at1.build.f_cpu=1000000L
attiny2313at1.build.core=tiny

###########################################################################

mschlenker, 2015-04-23:
The boards file looks OK, to cross check you might want to install: http://highlowtech.org/?p=1695 

Stay tuned. Give me another hour for the avrdude.conf

mschlenker, 2015-04-23:
What the hell? I just updated the AVR toolchain to 1.6.4 via the boards manager and tested with an ATtiny45 and the highlowtech version of the board definitions. The problem went away. As far as I can see there were no changes related to ATtiny45 (just to 85) in the avrdude.conf. If it goes away for you as well please close - this bug will probably resurface from time to time.

glprinamello, 2015-04-23:
still gives me:

Arduino:1.6.3 (Windows XP), Scheda:"ATtiny, ATtiny85, 1 MHz (internal)"

C:\Documents and
Settings\Giangy_Home\Desktop\PROGETTAZIONE\ArduinoProg\Arduino\hardware\tools\avr/bin/avrdude
-CC:\Documents and
Settings\Giangy_Home\Desktop\PROGETTAZIONE\ArduinoProg\Arduino\hardware\tools\avr/etc/avrdude.conf
-v -v -v -v -pattiny85 -cstk500v1 -PCOM2 -b19200 -e -Uefuse:w:0xff:m
-Uhfuse:w:0xdf:m -Ulfuse:w:0x62:m

avrdude: Version 6.0.1, compiled on Mar 30 2015 at 14:56:06

         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/

         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Documents and
Settings\Giangy_Home\Desktop\PROGETTAZIONE\ArduinoProg\Arduino\hardware\tools\avr/etc/avrdude.conf"

avrdude: error at C:\Documents and
Settings\Giangy_Home\Desktop\PROGETTAZIONE\ArduinoProg\Arduino\hardware\tools\avr/etc/avrdude.conf:332:
programmer type must be written as "id_type"

Errore durante la scrittura del bootloader

while trying to write boot to ATtiny85 1MHz internal

glprinamello, 2015-04-23:
where do i find the avr toolchain you are using?

mschlenker, 2015-04-23:
I am using the updated toolchain provided via the boards manager.

Would you try to replace your `C:\Documents and
Settings\Giangy_Home\Desktop\PROGETTAZIONE\ArduinoProg\Arduino\hardware\tools\avr/etc/avrdude.conf` with this file and test again:

https://gist.github.com/mschlenker/c3ac5c8bb455fc5f98df

BTW: Did you restart Arduino after doing "Tools > Boards > Boards Manager > Update"?

glprinamello, 2015-04-23:
Holy cow.. now it works.. why? you are great...

Thank you...

mschlenker, 2015-04-23:
@glprinamello Replacing `C:\Documents and Settings\Giangy_Home\Desktop\PROGETTAZIONE\ArduinoProg\Arduino\hardware\tools\avr/etc/avrdude.conf`did two things: First make sure the config is correct, second add the missing read section.

Da steh ich nun ich armer Tor und bin so klug als wie zuvor. 

Maybe the changes on avrdude did trigger some assumptions. I do not know, I probably never will. Please close this bug, it will resurface soon enough. Regarding the old avrdude.conf that was still used this might be another issue when instlled from zip.

glprinamello, 2015-04-23:
Thank you again have a nice day.

closed 2015-04-23 by glprinamello