arduino/Arduino #3024: IDE crashed on "Save"/"Save as..." on Ubuntu 14.10 with global menus (Jayatana)

neopostmodern, 2015-04-23:
*Setup: Ubuntu 14.10 x64, Java 1.7.0_80*

I tried the versions from the 
- Ubuntu Repositories
- latest stable from website 
- nightly from website 

...and all have the same behavior when trying to "Save" or "Save as...":

```bash
$ ./arduino 
Picked up JAVA_TOOL_OPTIONS: -javaagent:/usr/share/java/jayatanaag.jar 
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f68ec362324, pid=25817, tid=140085852890880
#
# JRE version: Java(TM) SE Runtime Environment (7.0_80-b15) (build 1.7.0_80-b15)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (24.80-b11 mixed mode linux-amd64 compressed oops)
# Problematic frame:
# C  [libpthread.so.0+0xa324]  pthread_mutex_lock+0x4
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/[...]/Downloads/arduino-nightly/hs_err_pid25817.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#
./arduino: line 29: 25817 Aborted                 (core dumped) java -Dswing.defaultlaf=com.sun.java.swing.plaf.gtk.GTKLookAndFeel $SPLASH processing.app.Base --curdir $CURDIR "$@"
```

Unsetting my JAVA_TOOL_OPTIONS (thus loosing the global menus) fixes this.

Any chance this could be fixed on the Arduino side? (Other Java programs work fine with Jayatana integrations).

ffissore, 2015-04-23:
Never heard of jayatana but it looks definitely the cause of your issue. What's your problem with global menus?

neopostmodern, 2015-04-23:
[Java Swing Jayatana](https://code.google.com/p/java-swing-ayatana/) _"is a project that brings Ubuntu global menu (AppMenu) and HUD support for Java Swing applications."_ (says [webupd8.org](http://www.webupd8.org/2014/02/get-unity-global-menu-hud-support-for.html))

It's obviously not strictly necessary but buys me the convenience of the HUD (searchable menus) and of course a nicer design.

I'm not sure this is on the Arduino side, but - as stated before - all other Java software I use runs just smooth.

mschlenker, 2015-04-23:
@neopostmodern Ubuntu 15.04 will soon be released - ubuntu.com still links to 14.10, but the release notes are available: https://wiki.ubuntu.com/VividVervet/ReleaseNotes Could you test whether the problem persists in 15.04? Since Ubuntu 14.10 is no LTS release, support will be ended soon.

ffissore, 2015-04-24:
jayatana looks interesting and a specific issue requesting its support would be great: please create such issue.
I'm going to close this however, since the problem you're experiencing is not related to the IDE but rather to something you did with it that we don't currently support

closed 2015-04-24 by ffissore

neopostmodern, 2015-04-24:
I haven't run the update yet. I'll create the specific jAyatana request once I have checked out 15.04

DanielWeigl, 2015-05-01:
Just FYI, you can start the Arduino IDE with following command to disable jayatanaag without uninstalling it

    JAVA_TOOL_OPTIONS="" arduino

SurferTim, 2015-05-07:
A feature request is ok for future releases. With Ubuntu 15.04, it doesn't work. Right now we needed a fix. DanielWeigl's solution works with all IDE versions and Ubunto 15.04. I edit the shell script.

mschlenker, 2015-05-07:
Wrapping the shell script is probably the best solution.

Even canonical is giving up on global menus. Don't know about HUD, thus. But Jayatana seems to be mostly abandonned. So why try riding a dead horse?

neopostmodern, 2015-05-07:
If you have more information on this, let me know, but from what I know they just *promoted* jayatana: http://www.webupd8.org/2015/01/ubuntu-1504-might-get-global-menu-and.html

And how and where is canonical giving up on global menus?

bashterm, 2015-06-03:
The solution presented on the arduino forums worked for me. http://forum.arduino.cc/index.php?topic=318520.0 is the link to the topic. You do lose the menus though

victorparragarcia, 2015-06-21:
Hello! I don't find any place to ask this :S The thing is that I'm using Ubuntu Vivid right now and the only software I have mandatory to use for work (Cafetran, a java app) is not working. It's weird cause it works on every other distros and Ubuntu flavours but just with Unity It crashes everytime the program opens the "java file manager" to pick a file to work with... Do you have any idea why this is happening?

ffissore, 2015-06-22:
Ask cafetran support