arduino/Arduino #3025: Feature Request: PWM read function

thijsvb, 2015-04-23:
It would be very usefull to have a function that can read a pwm signal from for example another arduino. Using the pulseIn function just isn't very practical, especially when trying to continuesly read a pin.

agdl, 2015-04-24:
What do you want to say exactly? Read the PWM and convert it back to a number from 0 to 255?

thijsvb, 2015-04-24:
Yes, exactly!

mikevanis, 2015-04-30:
I think this feature request wouldn't exist if the `pulseIn()` function was non-blocking. Just a thought!

facchinm, 2015-05-04:
You could check if #2409 solves your problems, as it contains an implementation of pulseIn based on micros() beside the normal implementation rewritten in ASM.
The pulseInLong() works correctly in interrupt context as it is interrupt based :smile:

agdl, 2015-05-07:
you can do it in a very simple way connecting a PWM output to an analog input by means of an RC filter (it exctract the average value of the PWM waveform)

agdl, 2015-05-22:
Waiting for feedback for too long. Reopen if needed

closed 2015-05-22 by agdl